YosysHQ/yosys #4119: Performance issue with many concatenations

flaviens, 2024-01-08:
### Version

Yosys 0.29+11 (git sha1 f7ffd12bf, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I noticed that for designs many wide concatenations, CXXRTL produces a C++ source that is extremely long to compile.
For example, I gave up on this [top.sv](https://github.com/YosysHQ/yosys/files/13861927/top_concats.zip) after more than an hour.

The commands are the usual as below, with an arbitrarily simple `tb.cxxrtl`:

`commands.sh`
```
yosys -p "read_verilog -sv top.sv; write_cxxrtl top.cpp"
g++ -g -O3 -std=c++14 -I `yosys-config --datdir`/include tb_cxxrtl.cc -o tb
./tb
```

The Yosys pass is incredibly fast, but g++ seems to take hours.
[Here is the resulting top.cpp](https://github.com/YosysHQ/yosys/files/13861972/cxxrtl.zip) if you would like to have a quick look.

Interestingly, all designs with similarly large and many concatenations do not seem to be affected equally by this performance issue.

I understand that this does not sound like the most pressing issue (and certainly isn't).

[Here](https://github.com/YosysHQ/yosys/files/13862036/top_longer.zip) is a longer design with the same observation.
I'm using `g++ (Ubuntu 10.5.0-1ubuntu1~20.04) 10.5.0`.

Thanks!

### Expected Behavior

g++ terminates in reasonable time (seconds or minutes)

### Actual Behavior

g++ runs for hours without an end in sight.

tomverbeure, 2024-01-08:
I’ve seen the same issue. Try it with clang++. It’s orders of magnitude faster on some template based code. And CXXRTL is nothing but templates.

whitequark, 2024-01-08:
Yeah, this is basically expected. I think this could be handled by splitting expressions into subexpressions when they reach a maximum depth.