YosysHQ/yosys #4122: Add a generic way to delete all non-synthesizable cells

nakengelhardt, 2024-01-09:
#4095 brings up an issue that we're going to run into more and more as we add non-synthesis features, but which I think we've always somewhat had with cells like `$assert` already: these should probably be ignored/deleted at some point in the `synth_*` passes rather than relying on the user to gate everything with `` `ifndef SYNTHESIS ``.

So that we don't have to keep editing all the synth passes, should we have a pass that deletes everything that doesn't make sense to have in the netlist passed to a P&R tool? Should it be a standalone pass or should we have an option to `delete` (or even just a new selection syntax specifier for non-synthesizable cells)?

povik, 2024-01-09:
How much do we expect the list of non-synthesizable instrumentation celltypes to grow? There’s the assertion/formal cells (with the planned addition of $check) and the print cells. Is there something else?

Could the tagging cells make their way into a synthesis flow? I am not familiar with those

In any case adding a selection specifier to pick out these special cells sounds like the best approach to me.

nakengelhardt, 2024-01-09:
Looking at the last years, "by a few cell types a year" seems a likely estimate. Currently planned there's also the addition of `$property`.

Selection specifier is probably the most flexible, and also a good occasion to finally add that selection specifier for "all (internal) FF cells" that we've been wanting to add for ages.

mattvenn, 2024-10-25:
Happens with $display and this has recently broken the asic flow

tcal-x, 2024-10-25:
Hi @mattvenn , for this specific case I add this: `yosys delete {*/t:$print}`

In context: https://github.com/hdl/bazel_rules_hdl/blob/main/synthesis/synth.tcl#L66-L69

KrystalDelusion, 2025-01-30:
During the changes I made for cell docs generation I added `bool is_synthesizable` (and `bool is_combinatorial`) to the CellType struct.  They are both assigned to a default value of `false` in the call to `CellTypes::setup_type()`, but if someone went through the list of cells and set the value to `true` where applicable then we could use that flag for deleting the non-synthesizable cells.