YosysHQ/yosys #4136: Yosys produces wrong cxxrtl model for some circuits

flaviens, 2024-01-18:
### Version

Yosys 0.37+1 (git sha1 242ae4ef0, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

Thank you very much with your so many contributions!
On this simple example repo that I could reduce, Yosys fails at producing a correct CXXRTL model (it gets produces but g++ complains).
I'm using `g++ (Ubuntu 10.5.0-1ubuntu1~20.04) 10.5.0` on Ubuntu 22.04.
Is there anything that I am doing wrongly?

Thank you!
Flavien

### Expected Behavior

Successful compilation

### Actual Behavior

Compilation fails

whitequark, 2024-01-18:
There is no example repo linked.

flaviens, 2024-01-18:
Sorry :disappointed: , edited the post

whitequark, 2024-01-19:
No, you did not do anything wrong. I broke one of the CXXRTL output modes in the `master` branch.

whitequark, 2024-01-19:
Actually, everything works for me when I build the `master` branch, and `performer` is not a part of the 0.37 release. I believe the problem is on your side.

whitequark, 2024-01-19:
I think you need to completely uninstall your Yosys reinstallation, rebuild Yosys from scratch from a fresh repo and reinstall it.

flaviens, 2024-01-20:
Indeed it is fixed now. Thank you very much @whitequark for your support!

closed 2024-01-20 by flaviens