YosysHQ/yosys #4144: Bug in the preparation of the simulation of some stateful cells in some cases

flaviens, 2024-01-20:
### Version

Yosys 0.37+21 (git sha1 8649e3066, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!
In some cases with stateful cells, the compilation step fails.
[I created a dedicated repository](https://github.com/flaviens/cxxrtl-b2) to help you reproduce this issue.
It does not seem related to #4136 , I am working with the last Yosys version.

Thank you!
Flavien

### Expected Behavior

Successful compilation

### Actual Behavior

```
cxxrtl_obj_dir/cxxrtl.cpp:118:90: error: ‘struct cxxrtl::slice_expr<cxxrtl::value<2>, 0, 0>’ has no member named ‘update’
  118 |  p_top__i_2e_celloutsig__0z.next.slice<0>() = p_top__i_2e_celloutsig__0z.next.slice<0>().update(value<1>{0x1u}, (p_top__i_2e_clkin__data.slice<96>().val() ? value<1>{0u} : (p_top__i_2e_clkin__data.slice<64>().val() ? value<1>{0x1u} : value<1>{0u})));
      |                                                                                          ^~~~~~
cxxrtl_obj_dir/cxxrtl.cpp:119:90: error: ‘struct cxxrtl::slice_expr<cxxrtl::value<2>, 0, 0>’ has no member named ‘update’
  119 |  p_top__i_2e_celloutsig__0z.next.slice<0>() = p_top__i_2e_celloutsig__0z.next.slice<0>().update(value<1>{0u}, (p_top__i_2e_clkin__data.slice<96>().val() ? value<1>{0x1u} : value<1>{0u}));
```

whitequark, 2024-01-20:
> [I created a dedicated repository](https://github.com/flaviens/cxxrtl-b2) to help you reproduce this issue.

Thank you, this really helps.

At a glance, I do think you found a real issue, I'll take a look shortly.

whitequark, 2024-01-20:
I'm not able to reproduce this. How are you getting that `.slice<0>()` in the middle? I've just built commit https://github.com/YosysHQ/yosys/commit/8649e306682e44b7f3e1b869d1043041db00b13e and the generated code looks like this:

	p_top__i_2e_celloutsig__0z.next = p_top__i_2e_celloutsig__0z.next.update(value<1>{0x1u}, (p_top__i_2e_clkin__data.slice<96>().val() ? value<1>{0u} : (p_top__i_2e_clkin__data.slice<64>().val() ? value<1>{0x1u} : value<1>{0u})));
	p_top__i_2e_celloutsig__0z.next = p_top__i_2e_celloutsig__0z.next.update(value<1>{0u}, (p_top__i_2e_clkin__data.slice<96>().val() ? value<1>{0x1u} : value<1>{0u}));

which is built without problems by g++ on my system:

    g++ (Debian 12.2.0-14) 12.2.0

Are you using a patched Yosys somehow? How are you getting all of these errors that I consistently cannot reproduce?

flaviens, 2024-01-21:
Hi @whitequark , thank you for your response! Indeed with again deleting everything and reinstalling again from scratch, the issue disappears. I suspect that cxxrtl.h moves between versions in the share directory and the wrong one may be selected sometimes when moving versions without a very clean uninstall. Thank you again for having looked at it :pray:

flaviens, 2024-01-21:
Hi again Whitequark, my apologies for the back and forth.
I'm actually still stumbling into the exact same bug for a small variation of the `top.sv`, which I updated [in the same repository](https://github.com/flaviens/cxxrtl-b2) now. I confirmed with a fresh Docker container that this occurs with a fresh initial installation.

You can try it with this Dockerfile if this is convenient for you:

```
FROM ubuntu:20.04
ENV DEBIAN_FRONTEND=noninteractive

# Install apt dependencies
RUN apt-get update && apt-get install -y build-essential clang bison flex \
	libreadline-dev gawk tcl-dev libffi-dev git \
	graphviz xdot pkg-config python3 libboost-system-dev \
	libboost-python-dev libboost-filesystem-dev zlib1g-dev

RUN git clone https://github.com/YosysHQ/yosys.git && cd yosys && git checkout 8649e30 && make config-clang && make -j$(nproc) && make install

RUN echo salut
RUN git clone https://github.com/flaviens/cxxrtl-b2.git
RUN cd cxxrtl-b2 && bash run.sh
```

whitequark, 2024-01-21:
> which I updated [in the same repository](https://github.com/flaviens/cxxrtl-b2) now.

I can reproduce your issue with the updated repository. I will look at it later. I do want to underscore that it is completely unacceptable to repeatedly file fuzzer-discovered bugs which do not reproduce on the `master` branch or with other blatant issues (some of the earlier testbenches referenced signals that do not exist, etc) and if this trend continues I will deprioritize all of your issues as they require me to spend quite a bit of time triaging them that I could spend helping people whose practical designs are blocked on some issue.

flaviens, 2024-01-21:
Hi Whitequark, thank you for your response! I understand, and am also learning from this process. I will continue trying to make it as easy as possible for you. Please prioritize as suits you the best. Thanks!

whitequark, 2024-01-24:
Fixed by https://github.com/YosysHQ/yosys/pull/4157.

flaviens, 2024-01-24:
Very cool, thanks!

PR #4157 (2024-01-24): cxxrtl: fix typo in codegen for async set/clear
Fixes #4144.
closed 2024-01-24 by flaviens