YosysHQ/yosys #4145: Bug in the generation of the simulation sources with some stateful cells

flaviens, 2024-01-20:
### Version

Yosys 0.37+21 (git sha1 8649e3066, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!
In some cases with stateful cells, cxxrtl.cpp generation fails.
[I created a dedicated repository](https://github.com/flaviens/cxxrtl-b3) to help you reproduce this issue.
I it simply that sensitivity to multiple signals is not implemented in CXXRTL?
Interestingly, in #4144 , there is also multiple sensitive signals but this error does not appear there when generating cxxrtl.cpp.

Thank you!
Flavien

### Expected Behavior

Successful cxxrtl.cpp generation

### Actual Behavior

```
3.3.9. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\wrapper_cxxrtl.\top_i.celloutsig_0z' using process `\wrapper_cxxrtl.$flatten\top_i.$proc$top.sv:9$1'.
ERROR: Multiple edge sensitive events found for this signal!
```

whitequark, 2024-01-20:
I don't believe that this is valid synthesizable Verilog. At least it is not valid according to IEEE 1364.1-2002, and it's clearly not accepted by Yosys.

```sv
  always_ff @(negedge clkin_data[0], posedge clkin_data[64])
    if (clkin_data[96] || clkin_data[64]) celloutsig_0z[0] <= 1'b0;
```

whitequark, 2024-01-20:
Please add a step to your CXXRTL fuzzing pipeline where you attempt to get the code you input to CXXRTL through any of the Yosys standard synthesis pipelines. I believe this should be enough:

    yosys -p 'hierarchy -top top; prep' input.sv

If this command fails it is probably not a bug in CXXRTL.

flaviens, 2024-01-20:
Thank you @whitequark  for your fast response!
This || is a last reduction that I did for you, maybe was not a great idea indeed ^^' if you'd like to try now, you can probably get the same bug by removing the last line of the always_ff block of #4144 (I left the office and can try myself another day, at your convenience)

whitequark, 2024-01-20:
> if you'd like to try now, you can probably get the same bug by removing the last line of the always_ff block of #4144 (I left the office and can try myself another day, at your convenience)

That's also not synthesizable Verilog and it fails the test I gave in https://github.com/YosysHQ/yosys/issues/4145#issuecomment-1902191869.

flaviens, 2024-01-20:
Ok sorry! Then I'll follow up later :)