YosysHQ/yosys #4146: Wrong runtime value with CXXRTL, probably because an event is ignored

flaviens, 2024-01-21:
### Version

Yosys 0.37+21 (git sha1 8649e3066, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I noticed wrong runtime values in some asynchronous cases. It may be because events are ignored (I am not yet familiar with CXXRTL internals at all, so it could be something completely different ^^).
I created a dedicated repository with a very reduced case for you to try.

Please let me know if you need any more infomation.

Thank you!
Flavien 

### Expected Behavior

Output value 1

### Actual Behavior

Output value 0

KrystalDelusion, 2024-01-21:
Without any calls to `top.step()` I don't see how this _could_ work.
From `help cxxrtl`:
> Note that CXXRTL simulations, just like the hardware they are simulating, are subject to race conditions. If, in the example above, the user logic would run simultaneously with the rising edge of the clock, the design would malfunction.
Adding a `top.step()` between `in_data[0] = 1` and `Apply the negative edge`, and then another two before reading back the output makes this work.  (I couldn't tell you why it needs two steps for the output to be available, but doing that gets the expected output).

Also as a sidenote, this is very much _not_ asynchronous, data is loaded on the clock edge ergo it is synchronous.

whitequark, 2024-01-21:
Yeah, this never even runs the code that evaluates the design.

whitequark, 2024-01-22:
@flaviens Are you writing a paper on automatic bug discovery and attempting to pad it by filing issues against an open source project so that it looks better during peer review?

closed 2024-01-21 by whitequark

flaviens, 2024-01-22:
Hello! I fixed the repo with what you noted. @KrystalDelusion pointed out exactly the problem: the negative edge (positive edges also, apparently) requires two calls of `step()`.
Probably (not 100% sure), this deviation will normally not propagate through the circuit but will only appear when being immediately read out. I think this is somehow problematic but maybe is ok if documented as there is no standard for that I guess.
Thank you!
Flavien