YosysHQ/yosys #4154: `rename -wire` affects quality of results in OpenROAD

oharboe, 2024-01-24:
### Version

Yosys 0.36 (git sha1 8f07a0d84, gcc 13.2.0-4ubuntu3 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

1. unzip [rename-test.zip](https://github.com/YosysHQ/yosys/files/14037520/rename-test.zip)
2. Run `yosys -s test.tcl`
3. Examine before.* and after.* files, I would have expected only file names to change, but there is more stuff that is moving around.

As to the difference in quality of results in OpenROAD, https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1774. I don't know if this is because there is actually a difference in quality of results or because it is just a change in the initial conditions that changes the results for e.g. placement.

It would be nice if `write_verilog` wrote out the result in a canonical form such that a rename does not affect the initial conditions of the OpenROAD tools.

### Expected Behavior

No change in quality of results in OpenROAD as a consequence of renaming

### Actual Behavior

Change in result.

oharboe, 2024-01-24:
Even narrowing the rename to just flip-flops, will change the ordering of the flip-flops in write_verilog:

```
renames -wire t:*DFF*
select -clear
```

povik, 2024-01-24:
Related to #3713