YosysHQ/yosys #4158: Assert `arg->is_signed == sig.as_wire()->is_signed' failed

ldoolitt, 2024-01-24:
### Version

Yosys 0.37 (git sha1 a5c7f69ed, gcc 10.2.1-6 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

get test case
[chirp_driver_bug.txt](https://github.com/YosysHQ/yosys/files/14042077/chirp_driver_bug.txt)
and rename as chirp_driver_bug.v

At the yosys prompt
`read_verilog chirp_driver_bug.v`

### Expected Behavior

```
1. Executing Verilog-2005 frontend: chirp_driver_bug.v
Parsing Verilog input from `chirp_driver_bug.v' to AST representation.
Generating RTLIL representation for module `\chirp_driver_bug'.
Successfully finished Verilog frontend.
```

I can get this behavior by removing the useless curly-braces on line 19.

### Actual Behavior

```
1. Executing Verilog-2005 frontend: chirp_driver_bug.v
Parsing Verilog input from `chirp_driver_bug.v' to AST representation.
Generating RTLIL representation for module `\chirp_driver_bug'.
ERROR: Assert `arg->is_signed == sig.as_wire()->is_signed' failed in /redacted/yosys/frontends/ast/genrtlil.cc:2067.
```

zachjs, 2024-01-29:
Slightly more minimal example below. I think this affects both inputs and outputs.
```Verilog
module concat_mod;
    reg signed [3:0] inp;
    wire signed [3:0] out;
    flip_mod fm(
        .inp({inp}),
        .out({out})
    );
endmodule

module flip_mod(
    input [3:0] inp,
    output [3:0] out
);
    assign out = ~inp;
endmodule
```

It also happens with:
```
        .inp($unsigned(inp)),
```

zachjs, 2024-01-29:
I believe this should be added to the logic using `lookup_cell_module()` to add indirection and resolve the inconsistency/ambiguity. I haven't started on this yet, but I hope to sometime next month.