YosysHQ/yosys #4159: accept "inout" on interface modports

ldoolitt, 2024-01-25:
### Feature Description

As of yosys-0.37 at least, using "inout" in a modport yields
```
ERROR: syntax error, unexpected TOK_INOUT, expecting TOK_ID or TOK_INPUT or TOK_OUTPUT or ')'
```
Is this situation intentional or permanent?

ldoolitt, 2024-01-29:
I'm not a SystemVerilog expert.
I do know that inout modports are accepted by Vivado and Verilator.
A few of them showed up in my $DAYJOB codebase.
Hence this question.

nakengelhardt, 2024-02-01:
Yosys only has very partial SystemVerilog support, the current `read_verilog` codebase is just not good enough to get anywhere near full support, although @zachjs has been doing his best to extend it. There are various plugins and converters for SystemVerilog input, such as sv2v or surelog. By the way, today's [Yosys User Group](https://blog.yosyshq.com/p/yosys-users-group/) meeting is also about this topic, the PULP folks will be presenting their tool [svase](https://github.com/pulp-platform/svase).