YosysHQ/yosys #4164: Missynthesis of wide shifts

flaviens, 2024-01-26:
### Version

Yosys 0.37+29 (git sha1 3c3788ee2, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I stumbled into an issue where a wide shift gets some misoptimization.
[I could reduce it to a very simple circuit, that is packaged into this repository for helping you reproduce the issue](https://github.com/flaviens/yosys-b2).
Interestingly, reducing the shift operand width by 1 more will clear the bug.

Please let me know if you require any more information or help, or please let me know if I did something wrong here.

Thank you!
Flavien

### Expected Behavior

no optimization
Output: 00000000
with optimization
Output: 00000000

### Actual Behavior

no optimization
Output: 00000000
with optimization
Output: ffffffff

povik, 2024-01-26:
Looks like the blame is on `SigSpec::as_int` overflowing here
https://github.com/YosysHQ/yosys/blob/80511ced71b657e77a5389d54c6aa045c047ccaf/passes/opt/opt_expr.cc#L1306

jix, 2024-01-26:
This might be the same bug as #4010. I have some work in progress to detect and error out on `as_int` overflows in general together with some fixes for places where it could overflow (like this optimization) but I haven't found the time finish it.

georgerennie, 2025-06-04:
This is fixed in current main by #5101 or similar (there were a couple different patches fixing edge cases of the shifts).

closed 2025-06-04 by georgerennie