YosysHQ/yosys #4194: Add an option to synth to enable the booth lower power option

maliberty, 2024-02-08:
### Feature Description

The booth command has a -lowpower option which is quite useful.  The synth command only exposes -booth but with no way to set this option.  You can replace synth with its equivalent steps but that is quite inconvenient to just add  this one flag.

It would be great to have a -lowpower option (or -booth_lowpower) option on synth to simplify the usage.

precisionmoon, 2024-02-08:
Perhaps can we enable -booth lowpower by default and provide a switch to disable -booth (like -no_booth) or change booth setting from lowpower to fast?

povik, 2024-02-08:
> It would be great to have a -lowpower option (or -booth_lowpower) option on synth to simplify the usage.

For that it is useful to get a confirmation that the `booth -lowpower` architecture is useful for somebody's flows. I take it you are just giving that confirmation.

> Perhaps can we enable -booth lowpower by default and provide a switch to disable -booth (like -no_booth) or change booth setting from lowpower to fast?

Not sure how to decide this, but at least one issue preventing us from making `booth -lowpower` the default for anything is that it maps signed multipliers only, so we would need to address that first.

maliberty, 2024-02-08:
The reported gains are enough to make it interesting to the OpenROAD team.  We are evaluating how to enable it in our flow.

I believe the original implementation was that there were separate signed/unsigned options but when the PR came into yosys it became default/lowpower.   Perhaps it would be possible to restore some of the original intent as an alternative to this flag?

@andyfox-rushc any comments?

povik, 2024-02-08:
> I believe the original implementation was that there were separate signed/unsigned options but when the PR came into yosys it became default/lowpower. Perhaps it would be possible to restore some of the original intent as an alternative to this flag?

The original implementation had what is now `booth` for mapping unsigned multipliers, and what is now `booth -lowpower` for mapping signed ones. So the properties of the mapped multiplier you got depended on its signedness, which we deemed undesirable.

What we did is we generalized the unsigned architecture, and made it the `booth` default. What was there originally to map signed multipliers, we hid under the `-lowpower` option, but never got to generalize it.

If what it takes to generalize the `-lowpower` architecture is to pad the operands in the unsigned case, then that is simple enough, we can do that immediately. I am just not sure if there isn't some better, clever way to do it, in which case someone needs to focus in on that.

precisionmoon, 2024-02-08:
> > It would be great to have a -lowpower option (or -booth_lowpower) option on synth to simplify the usage.
> 
> For that it is useful to get a confirmation that the `booth -lowpower` architecture is useful for somebody's flows. I take it you are just giving that confirmation.

Sorry, no not yet.   We're still doing validation.   We'll let you know if we see consistently better area and delay across most designs.

povik, 2024-02-08:
> We'll let you know if we see consistently better area and delay across most designs.

OK

maliberty, 2024-02-08:
My understanding is that they are two substantively different architectures and that there was a good benefit from using the lowpower/signed versus generalizing the default/unsigned case.   Hopefully Cho's experiments will demonstrate the same.