YosysHQ/yosys #4214: Support for GAL and ATF-series SPLDs

LiamM32, 2024-02-15:
### Feature Description

Add support for the old GAL PLDs and their modern equivalent; ATF-series PLDs such as the ATF16V8C.

whitequark, 2024-02-15:
If you want to see this happen you will need to work on this yourself. For example, you could use the information provided by [Project Bureau](https://whitequark.github.io/prjbureau/) to build a synthesizer and a fitter for ATF1502AS/BE.

sporniket, 2024-08-22:
Other reference materials that use yosys "plus something else" to get things done : 
* https://github.com/annoyatron255/yosys4gal : yosys --> adhoc pnr --> jedec file ; python scripts
* https://github.com/psurply/ReGAL : yosys --> galette --> jedec file ; tcl scripts and rust programs