YosysHQ/yosys #4217: It seems that a compilation syntax error occurred with yosys

yelen103, 2024-02-17:
### Version

 Yosys 0.37+1 (git sha1 e1f4c5c9cbb, clang  -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Get the GitHub link from the documentation and run the Quartus project direc
[yosys.docx](https://github.com/YosysHQ/yosys/files/14318068/yosys.docx)
tly

### Expected Behavior

Generate files with no syntax errors via the 'verilog_write' command

### Actual Behavior

There was a syntax error - the same variable was assigned in different always, and the asynchronous trigger was not merged

whitequark, 2024-02-17:
@yelen103 Could you please attach the code that reproduces the issue as a file?

yelen103, 2024-02-18:
[sysverilog_quartus_yosys.zip](https://github.com/YosysHQ/yosys/files/14320964/sysverilog_quartus_yosys.zip)
thank you