YosysHQ/yosys #4225: cxxrtl hangs on a loop

gatecat, 2024-02-21:
### Version

Yosys 0.38+75 (git sha1 01d6c12af, gcc 13.2.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Make some Verilog with a typo like `assign x = x;`

```verilog
module top;
assign x = x;
endmodule
```

```
yosys -p "hierarchy -top top; write_cxxrtl" loop.v
```

### Expected Behavior

Either it can be implemented with delta cycles, or an error.

### Actual Behavior

write_cxxrtl hangs forever after:
```
Module `top' contains feedback arcs through wires:
  x
```

povik, 2024-02-21:
For a loop like this to be caught by `check` we need the change of behavior from #3965.

whitequark, 2024-02-21:
> Either it can be implemented with delta cycles, or an error.

Yeah, I agree.

whitequark, 2024-02-21:
> For a loop like this to be caught by `check` we need the change of behavior from #3965.

I think we don't necessarily have to require `check` here, this should be solvable on CXXRTL side.