YosysHQ/yosys #4227: CXXRTL: assert '!for_debug' failed with `write_cxxrtl -g4`

jfng, 2024-02-21:
### Version

Yosys 0.38+88 (git sha1 d5934357f, gcc 13.2.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Create the following `repro.il` file:
```
attribute \cxxrtl_blackbox 1
attribute \blackbox 1
module \jtag_model
  wire output 3 \tms
end

module \sim_top
  wire \jtag_tms
  cell \jtag_model \U$0
    connect \tms \jtag_tms
  end
end
```

Run `yosys repro.il -p "write_cxxrtl -g4"`

### Expected Behavior

A CXXRTL output.

### Actual Behavior

```
ERROR: Assert `!for_debug' failed in backends/cxxrtl/cxxrtl_backend.cc:1522.
```

jfng, 2024-02-21:
This error does *not* occur with:
- any `-O` level below 5 (e.g. `-O4 -g4`)
- any `-g` level below 4 (e.g. `-O6 -g3`)

whitequark, 2024-02-21:
That looks like a bug in the repro as you're connecting a 3-bit port to a 1-bit wire without any adjustment.

povik, 2024-02-21:
> That looks like a bug in the repro as you're connecting a 3-bit port to a 1-bit wire without any adjustment.

The 3 happens to refer to the port ID, not the width of the port.

whitequark, 2024-02-21:
Oh. I really don't like RTLIL U_U

whitequark, 2024-02-21:
I'm not actually sure whether this is valid input for the CXXRTL backend as-is (it might be over-minimized?) but I'd have to investigate why it fails and think about the semantics of this.

gatecat, 2024-02-27:
The assert was removed in https://github.com/YosysHQ/yosys/commit/d903f47d41cad38ab80c74095ece198afb947ae7, not sure if the issue is still relevant.