YosysHQ/yosys #4231: Verilog asynchronous reset pattern leads to corrupted `$check`, `$print` control signals

povik, 2024-02-23:
### Version

Yosys 0.38+92 (git sha1 f7737a12c, clang++ 11.1.0 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

```
read_verilog -sv <<EOF

module top(q, d, clk, rst);
	input wire clk;
	input wire rst;
	input wire [1:0] d;
	output reg [1:0] q;
	always @(posedge clk, posedge rst) begin
		if (rst) begin
			q <= 0;
		end else begin
			q <= d;
			$display("q <= %x", d);
			assert(!(d & (d - 1)));
		end
	end
endmodule

EOF
proc
dump t:$print t:$check
```

### Expected Behavior

The emitted `$print` and `$check` cells are conditioned on `!rst`.

### Actual Behavior

The `EN` port of the cells in question is driven high by a constant.

``` 
  attribute \src "<<EOF:13.4-13.26"
  cell $check $assert$<<EOF:13$3
    parameter \ARGS_WIDTH 0
    parameter \FORMAT ""
    parameter \PRIORITY 32'11111111111111111111111111111110
    parameter \TRG_POLARITY 2'11
    parameter \TRG_ENABLE 1
    parameter \TRG_WIDTH 2
    parameter \FLAVOR "assert"
    connect \ARGS { }
    connect \A $logic_not$<<EOF:13$6_Y
    connect \EN 1'1
    connect \TRG { \rst \clk }
  end

  attribute \src "<<EOF:12.4-12.27"
  cell $print $display$<<EOF:12$2
    parameter \ARGS_WIDTH 2
    parameter \FORMAT "q <= {2:>01hu}\n"
    parameter \PRIORITY 32'11111111111111111111111111111111
    parameter \TRG_POLARITY 2'11
    parameter \TRG_ENABLE 1
    parameter \TRG_WIDTH 2
    connect \ARGS \d
    connect \EN 1'1
    connect \TRG { \rst \clk }
  end
```

whitequark, 2024-02-23:
> AFAICT this exact issue of there being hidden assumptions on RTLIL processes is what prompted Amaranth to give up on emitting synchronization rules entirely: [amaranth-lang/amaranth#603](https://github.com/amaranth-lang/amaranth/issues/603)

Yep. (It's also that it is a completely undocumented contract between two parts of Yosys, which Amaranth really should not be relying on.)

My view is that Verilog frontend should stop emitting them too, by means similar to those of Amaranth, and then sync rules should be removed from RTLIL.

povik, 2024-02-23:
> My view is that Verilog frontend should stop emitting them too, by means similar to those of Amaranth, and then sync rules should be removed from RTLIL.

One obstacle to making this change is that the synchronization rules are a useful intermediate representation for the inference of latches.

whitequark, 2024-02-23:
Can you not do this while collecting left-hand sides of assignments? If every branch assigns to a bit, then that is a flop, otherwise a latch.

povik, 2024-02-23:
That would still regress this example, though I have no idea whether this is idiomatic Verilog and something we ought to support
```
module top(q, d, en);
	input wire en;
	input wire [1:0] d;
	output reg [1:0] q;
	always @(*) begin
		if (en)
			q <= d;
		else
			q <= q;
	end
endmodule
```

Of course we could detect when the LHS and RHS bit are identical, and special-handle those, but then there are more complicated examples that are covered under the current traversal of the mux network by `proc_dlatch`. In any case we could introduce a special cell to be consumed by `proc_dlatch`, independent on any processes, and replace the current scheme with that.

whitequark, 2024-02-23:
> That would still regress this example, though I have no idea whether this is idiomatic Verilog and something we ought to support

IEEE 1364.1 doesn't consider this a latch:

[image]

So I think we should not care about it. No one really wants latches anyway.

povik, 2024-02-23:
That makes things simpler. So a combinational loop would be emitted for that example if we move the inference of latches up into the frontend.

whitequark, 2024-02-23:
Yep!