YosysHQ/yosys #4237: ABC9/XAIGER (synth_ecp5) crash "Executing AIGER Frontend" std::vector<_Tp, _Alloc>::....  Assertion '__n < this->size()' failed.

mangelajo, 2024-02-25:
### Version

Yosys 0.38+4 (git sha1 ac0fb2e3011, gcc 14.0.1 -O2 -fexceptions -fstack-protector-strong -mbranch-protection=standard -fasynchronous-unwind-tables -fstack-clash-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

# I narrowed it down to a specific module, with -noabc9 it works fine
git clone https://github.com/WangXuan95/FPGA-USB-Device
cd FPGA-USB-Device/
yosys <<EOF
read_verilog RTL/usbfs_core/usbfs_debug_uart_tx.v
synth_ecp5 -top usbfs_debug_uart_tx
EOF

### Expected Behavior

```
yosys> read_verilog RTL/usbfs_core/usbfs_debug_uart_tx.v
1. Executing Verilog-2005 frontend: RTL/usbfs_core/usbfs_debug_uart_tx.v
Parsing Verilog input from `RTL/usbfs_core/usbfs_debug_uart_tx.v' to AST representation.
Generating RTLIL representation for module `\usbfs_debug_uart_tx'.
Successfully finished Verilog frontend.

yosys> synth_ecp5 -top usbfs_debug_uart_tx

....

2.42.19.3. Executing XAIGER backend.
<suppressed ~11 debug messages>
Extracted 328 AND gates and 1220 wires from module `usbfs_debug_uart_tx' to a netlist network with 114 inputs and 180 outputs.

2.42.19.4. Executing ABC9_EXE pass (technology mapping using ABC9).

2.42.19.5. Executing ABC9.
Running ABC command: "<yosys-exe-dir>/abc" -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_lut <abc-temp-dir>/input.lut
ABC: + read_box <abc-temp-dir>/input.box
ABC: + &read <abc-temp-dir>/input.xaig
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =    114/    180  and =     253  lev =   10 (0.54)  mem = 0.02 MB  box = 110  bb = 67
ABC: + &scorr
ABC: Warning: The network is combinational.
ABC: + &sweep
ABC: + &dc2
ABC: + &dch -f
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =    114/    180  and =     284  lev =   10 (0.34)  mem = 0.02 MB  ch =   23  box = 99  bb = 67
ABC: + &if -W 300 -v
ABC: K = 7. Memory (bytes): Truth =    0. Cut =   60. Obj =  140. Set =  636. CutMin = no
ABC: Node =     284.  Ch =    21.  Total mem =    0.17 MB. Peak cut mem =    0.01 MB.
ABC: P:  Del = 2096.00.  Ar =     196.0.  Edge =      279.  Cut =     2653.  T =     0.00 sec
ABC: P:  Del = 2096.00.  Ar =     196.0.  Edge =      280.  Cut =     2621.  T =     0.00 sec
ABC: P:  Del = 2096.00.  Ar =     166.0.  Edge =      271.  Cut =     3677.  T =     0.00 sec
ABC: F:  Del = 2089.00.  Ar =     104.0.  Edge =      256.  Cut =     2943.  T =     0.00 sec
ABC: A:  Del = 2089.00.  Ar =     104.0.  Edge =      254.  Cut =     3019.  T =     0.00 sec
ABC: A:  Del = 2089.00.  Ar =     104.0.  Edge =      254.  Cut =     3015.  T =     0.00 sec
ABC: Total time =     0.00 sec
ABC: + &write -n <abc-temp-dir>/output.aig
ABC: + &mfs
ABC: + &ps -l
ABC: <abc-temp-dir>/input : i/o =    114/    180  and =     226  lev =   10 (0.29)  mem = 0.02 MB  box = 99  bb = 67
ABC: Mapping (K=7)  :  lut =     53  edge =     209  lev =    4 (0.12)  Boxes are not in a topological order. Switching to level computation without boxes.
ABC: levB =   10  mem = 0.01 MB
ABC: LUT = 53 : 2=4 7.5 %  3=16 30.2 %  4=16 30.2 %  5=15 28.3 %  6=0 0.0 %  7=2 3.8 %  Ave = 3.94
ABC: + &write -n <abc-temp-dir>/output.aig
ABC: + time
ABC: elapse: 0.03 seconds, total: 0.03 seconds

2.42.19.6. Executing AIGER frontend.
/usr/include/c++/14/bits/stl_vector.h:1127: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t; _Alloc = std::allocator<Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t>; reference = Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)

```

### Actual Behavior

No Crash

Ravenslofty, 2024-03-03:
My build doesn't have a lot of the defensive flags that yours does, but I can't reproduce this on `Yosys 0.38+113 (git sha1 91fbd5898, clang++ 16.0.6 -fPIC -Os)`.

mangelajo, 2024-03-07:
I am going to compile from master and test. But I suspect that the extra-defensive flags used in fedora could be catching an actual bug.

mangelajo, 2024-03-07:
I recompiled with latest version, and also switched the fedora spec file to use the internal abc9 instead of external abc9, but same crash:

```
Starting program: /usr/bin/yosys -q -s "bin/fpga_top_usb_camera.ys"

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.fedoraproject.org/>
Enable debuginfod for this session? (y or [n]) n
Debuginfod has been disabled.
To make this setting permanent, add 'set debuginfod enabled off' to .gdbinit.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Warning: Yosys has only limited support for tri-state logic at the moment. (ip/FPGA-USB-Device/RTL/usbfs_core/usbfs_core_top.v:167)
Warning: Yosys has only limited support for tri-state logic at the moment. (ip/FPGA-USB-Device/RTL/usbfs_core/usbfs_core_top.v:168)
[Detaching after vfork from child process 140663]
/usr/include/c++/14/bits/stl_vector.h:1127: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t; _Alloc = std::allocator<Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t>; reference = Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.

Program received signal SIGABRT, Aborted.
0x0000fffff77c7c40 in __pthread_kill_implementation () from /lib64/libc.so.6
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.39.9000-3.fc41.aarch64 libffi-3.4.4-8.fc41.aarch64 libgcc-14.0.1-0.8.fc41.aarch64 libstdc++-14.0.1-0.8.fc41.aarch64 ncurses-libs-6.4-12.20240127.fc40.aarch64 readline-8.2-8.fc40.aarch64 tcl-8.6.14-1.fc41.aarch64 zlib-ng-compat-2.1.6-2.fc40.aarch64
(gdb) bt
#0  0x0000fffff77c7c40 in __pthread_kill_implementation () from /lib64/libc.so.6
#1  0x0000fffff7775940 [PAC] in raise () from /lib64/libc.so.6
#2  0x0000fffff7760288 [PAC] in abort () from /lib64/libc.so.6
#3  0x0000fffff7cd20a8 [PAC] in std::__glibcxx_assert_fail(char const*, int, char const*, char const*) () from /lib64/libstdc++.so.6
#4  0x0000000000583c80 [PAC] in std::vector<Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::entry_t, std::allocator<Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::entry_t> >::operator[] (this=<optimized out>, __n=<optimized out>)
    at /usr/include/c++/14/bits/stl_vector.h:1127
#5  std::vector<Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::entry_t, std::allocator<Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::entry_t> >::operator[] (this=<optimized out>, __n=<optimized out>)
    at /usr/include/c++/14/bits/stl_vector.h:1125
#6  0x000000000058b91c [PAC] in Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::iterator::operator-> (
    this=<synthetic pointer>) at ./kernel/hashlib.h:419
#7  Yosys::ConstEvalAig::set_incremental (this=0xffffffffdc68, sig=..., value=...) at frontends/aiger/aigerparse.cc:121
#8  Yosys::AigerReader::parse_xaiger (this=this@entry=0xffffffffdea8) at frontends/aiger/aigerparse.cc:446
#9  0x000000000058c69c [PAC] in Yosys::AigerFrontend::execute (this=this@entry=0xcc1470 <Yosys::AigerFrontend>, f=@0xffffffffe048: 0x10104b30,
    filename="/tmp/yosys-abc-1qBiO6/output.aig", args=std::vector of length 8, capacity 8 = {...}, design=design@entry=0xd1f510) at frontends/aiger/aigerparse.cc:1062
#10 0x0000000000484ed0 [PAC] in Yosys::Frontend::execute (this=this@entry=0xcc1470 <Yosys::AigerFrontend>, args=std::vector of length 8, capacity 8 = {...}, design=design@entry=0xd1f510)
    at kernel/register.cc:469
#11 0x00000000004840a8 [PAC] in Yosys::Pass::call (design=design@entry=0xd1f510, args=std::vector of length 8, capacity 8 = {...}) at kernel/register.cc:317
#12 0x00000000004844f4 [PAC] in Yosys::Pass::call (design=design@entry=0xd1f510,
    command="read_aiger -xaiger -wideports -module_name fpga_top_usb_camera$abc9 -map /tmp/yosys-abc-1qBiO6/input.sym /tmp/yosys-abc-1qBiO6/output.aig") at kernel/register.cc:294
#13 0x0000000000484ae0 [PAC] in Yosys::ScriptPass::run_nocheck (this=<optimized out>, command=..., info=...) at kernel/register.cc:416
#14 0x00000000008a63d8 [PAC] in (anonymous namespace)::Abc9Pass::script (this=0xcb3728 <(anonymous namespace)::Abc9Pass>) at passes/techmap/abc9.cc:439
#15 0x00000000008a1af8 [PAC] in (anonymous namespace)::Abc9Pass::execute (this=this@entry=0xcb3728 <(anonymous namespace)::Abc9Pass>, args=std::vector of length 3, capacity 3 = {...},
    design=design@entry=0xd1f510) at passes/techmap/abc9.cc:276
#16 0x00000000004840a8 [PAC] in Yosys::Pass::call (design=design@entry=0xd1f510, args=std::vector of length 3, capacity 3 = {...}) at kernel/register.cc:317
#17 0x00000000004844f4 [PAC] in Yosys::Pass::call (design=design@entry=0xd1f510, command="abc9 -W 300") at kernel/register.cc:294
#18 0x00000000004849ac [PAC] in Yosys::ScriptPass::run (this=0xcbb430 <(anonymous namespace)::SynthEcp5Pass>, command=..., info=...) at kernel/register.cc:403
#19 0x00000000009c796c [PAC] in (anonymous namespace)::SynthEcp5Pass::script (this=0xcbb430 <(anonymous namespace)::SynthEcp5Pass>) at techlibs/ecp5/synth_ecp5.cc:391
[... cut ...]
#21 0x00000000004840a8 [PAC] in Yosys::Pass::call (design=design@entry=0xd1f510, args=std::vector of length 3, capacity 3 = {...}) at kernel/register.cc:317
#22 0x00000000004844f4 [PAC] in Yosys::Pass::call (design=0xd1f510, command=...) at kernel/register.cc:294
#23 0x00000000004f19bc [PAC] in Yosys::run_frontend (filename="bin/fpga_top_usb_camera.ys", command=..., design=0xd1f510, from_to_label=<optimized out>) at kernel/yosys.cc:1200
#24 0x0000000000454e68 [PAC] in main (argc=4, argv=0xfffffffff258) at kernel/driver.cc:642
```

mangelajo, 2024-03-07:
For completeness I also tried compiling with clang:

```$ yosys --version
Yosys 0.38+129 (git sha1 e9cd6ca9e8f, clang++ 18.1.0 -O2 -flto=auto -ffat-lto-objects -fexceptions -fstack-protector-strong -mbranch-protection=standard -fasynchronous-unwind-tables -fstack-clash-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fPIC -Os) ```

mangelajo, 2024-03-08:
> My build doesn't have a lot of the defensive flags that yours does, but I can't reproduce this on `Yosys 0.38+113 (git sha1 91fbd5898, clang++ 16.0.6 -fPIC -Os)`.

I compiled in gcc without all the defensive flags,

```
$ ./yosys --version
Yosys 0.38+129 (git sha1 e9cd6ca9e8f, g++ 14.0.1 -fPIC -O0)
```

 but, even if it´s not crashing for the out-of-bounds access, it fails at the end with:

```
2.42.19.6. Executing AIGER frontend.
<suppressed ~611 debug messages>
Removed 362 unused cells and 1254 unused wires.

2.42.19.7. Executing ABC9_OPS pass (helper functions for ABC9).
ABC RESULTS:              $lut cells:       57
ABC RESULTS:   $paramod$37e41664cab9052a9b41c89036e0524f666a0fee\TRELLIS_FF_$abc9_byp cells:       26
ABC RESULTS:   $paramod$68581f66a8fa831820fa7633bd24ba569ae9d7e3\TRELLIS_FF_$abc9_byp cells:       32
ABC RESULTS:   $paramod$838872d5a4bab89607f53482b205c0fd50d8b82e\CCU2C cells:       32
ABC RESULTS:   $paramod$b50ed6cf3f2b5c0e07657602cb4455d90f2562f9\TRELLIS_FF_$abc9_byp cells:        9
ERROR: Assert `wire' failed in passes/techmap/abc9_ops.cc:1455.
```

Ravenslofty, 2024-03-23:
Wait, `-mbranch-protection=standard`? Are you running on ARM64?