YosysHQ/yosys #4242: Incorrect BitVector width in SMT2 code produced by yosys

pietrobe03, 2024-02-27:
### Version

Yosys 0.37 (git sha1 a5c7f69ed8f, gcc 13.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -ffile-prefix-map=/build/yosys/src=/usr/src/debug/yosys -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

The bug can be reproduced using the following example module: [TwoBufferExample.zip](https://github.com/YosysHQ/yosys/files/14418935/TwoBufferExample.zip).

1. Within yosys, I load the Verilog file using `read_verilog TwoBuffer.v`.
2. I apply `proc` to to the loaded modules.
3. Using `write_smt2 TwoBuffer.smt2`, I generate a SMT2 description of the modules.

### Expected Behavior

The SMT2 description generated by yosys should contain correct type annotations, such that a solver like z3 can interpret it without any errors. For the given example, the declaration of `|TwoBuffer#4|` (representing `r1.o` within the `TwoBuffer` module) should have a result type of `(_ BitVec 32)`. A full version of the expected SMT2 output can be found in the linked example.

### Actual Behavior

The SMT2 description generated by yosys contains an incorrect bitwidth for the `o` wire in cell `\r1` within the `TwoBuffer` module. In the declaration of function `|TwoBuffer#4|`, the produced result type is `(_ BitVec 1)` instead of the expected `(_ BitVec 32)`. When interpreting that SMT2 output with z3, the following error appears. 
```
(error "line 68 column 68: Sorts (_ BitVec 1) and (_ BitVec 32) are incompatible")
```
A full version of the actual SMT2 output can be found in the linked example.

pietrobe03, 2024-02-27:
Thank you for the quick and helpful response! I'm fairly new to Verilog and circuit verification, so that's actually a misunderstanding on my side. Also thank you for the remark on notation!

closed 2024-02-27 by pietrobe03

povik, 2024-02-27:
> so writing `r2.i` and `r1.o` doesn't actually reference those signals from the other module. Instead, it implicitly creates new nets (an unfortunate feature of verilog), which are defaulted to a width of 1 here.

That sounds like a Yosys bug, because the Verilog standard doesn't say we should be creating implicit wires for identifiers which look like hierarchical references, so we should error out on those occurrences, I think.