YosysHQ/yosys #4244: Cannot build yosys in Ubuntu following instructions in README

RCoeurjoly, 2024-02-27:
### Version

N/A

### On which OS did this happen?

Linux

### Reproduction Steps

```
sudo apt-get install build-essential clang bison flex \
	libreadline-dev gawk tcl-dev libffi-dev git \
	graphviz xdot pkg-config python3 libboost-system-dev \
	libboost-python-dev libboost-filesystem-dev zlib1g-dev
make config-clang
make
```

### Expected Behavior

Build successful.

### Actual Behavior

In file included from kernel/satgen.cc:20:
In file included from ./kernel/satgen.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/calc.cc:24:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/rtlil.cc:20:
./kernel/yosys.h:42:10: fatal error: In file included from kernel/cellaigs.cc:'map' file not found20
:
In file included from ./kernel/cellaigs.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>#include <map>

         ^~~~~         ^~~~~

In file included from kernel/log.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/driver.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/celledges.cc:20:
In file included from ./kernel/celledges.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/binding.cc:20:
In file included from kernel/binding.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  5%] Building kernel/json.o
In file included from kernel/yosys.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
In file included from kernel/qcsat.cc:20:
In file included from ./kernel/qcsat.h:23:
In file included from ./kernel/satgen.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
make: *** [Makefile:748: kernel/celledges.o] Error 1
make: *** Waiting for unfinished jobs....
1 error generated.
make: *** [Makefile:748: kernel/driver.o] Error 1
In file included from kernel/mem.cc:20:
In file included from ./kernel/mem.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
make: *** [Makefile:748: kernel/cellaigs.o] Error 1
In file included from kernel/scopeinfo.cc:20:
./kernel/scopeinfo.h:23:10: fatal error: 'vector' file not found
#include <vector>
         ^~~~~~~~
1 error generated.
make: *** [Makefile:748: kernel/binding.o] Error 1
In file included from kernel/ffmerge.cc:20:
In file included from ./kernel/ffmerge.h:23:
In file included from ./kernel/ffinit.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/yw.cc:20:
In file included from ./kernel/yw.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/ff.cc:20:
In file included from ./kernel/ff.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/json.cc:20:
In file included from ./kernel/json.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
1 error generated.
make: *** [Makefile:748: kernel/satgen.o] Error 1
1 error generated.
make: *** [Makefile:748: kernel/calc.o] Error 1
make: *** [Makefile:748: kernel/log.o] Error 1
1 error generated.
make: *** [Makefile:748: kernel/mem.o] Error 1
1 error generated.
make: *** [Makefile:748: kernel/json.o] Error 1
1 error generated.
1 error generated.
make: *** [Makefile:748: kernel/ff.o] Error 1
make: *** [Makefile:748: kernel/qcsat.o] Error 1
1 error generated.
1 error generated.
1 error generated.
make: *** [Makefile:748: kernel/yw.o] Error 1
make: *** [Makefile:748: kernel/yosys.o] Error 1
make: *** [Makefile:748: kernel/ffmerge.o] Error 1
1 error generated.
make: *** [Makefile:748: kernel/scopeinfo.o] Error 1
1 error generated.
make: *** [Makefile:748: kernel/rtlil.o] Error 1

RCoeurjoly, 2024-02-27:
Note that `make CXX=$CXX` builds yosys correctly.

Ravenslofty, 2024-02-28:
Those are C++ standard headers that it can't find; I think there might be a problem with your compiler.

spth, 2024-02-29:
I see the same problem on amd64 Debian GNU/Linux testing for the yosys 0.38 release. My LLVM/clang++ works fine when compiling other C++ programs, and compiling yosys using the config-gcc also works.

So I think this is a bug in the config-clang target in the yosys build system.

Is there some way to make the yosys build system show  the actual commands used to invoke the C++ compiler?

KrystalDelusion, 2024-03-03:
> Is there some way to make the yosys build system show the actual commands used to invoke the C++ compiler?

Try `make PRETTY=0`

KrystalDelusion, 2024-03-03:
> Note that `make CXX=$CXX` builds yosys correctly.

I suspect this is because all of the $(CONFIG) checks are along the lines of
```
ifeq ($(CONFIG),clang)
CXX = clang++
```
which means they won't use the environment variable by default.
Changing all of them to (e.g.) `CXX ?= clang++` should fix this?

ripudamank2, 2024-03-17:
is this problem is solved ? I am getting the same error.

KrystalDelusion, 2024-03-17:
@ripudamank2 the fix has not been merged yet, have you tried the `make CXX=$CXX` workaround?

ripudamank2, 2024-03-18:
While installing via ORFS, when downloading the Yosys package and attempting to execute the "make" command, it becomes stuck due to the error mentioned above.

KrystalDelusion, 2024-03-18:
According to [their documentation](https://openroad-flow-scripts.readthedocs.io/en/latest/#build-command-and-options) you should be able to use `--yosys-args STRING` when calling `./build_openroad.sh`, but if you're still having the same issue as the thread you had over there where you're not able to get past `setup.sh` then that probably doesn't help.

If you want help for Yosys specifically I suggest trying to follow the [Yosys instructions](https://github.com/YosysHQ/yosys?tab=readme-ov-file#building-from-source) instead of the open road instructions.

ripudamank2, 2024-03-19:
I was using ORFS  few weeks ago but last week i deleted the ORFS and done git pull again . Its getting stuck in that error .

KrystalDelusion, 2024-03-20:
I have version 10 locally, the CI tests run regularly against clang-11, clang-12, and clang-15

ripudamank2, 2024-03-20:
Can you try version 14 of clang ?

KrystalDelusion, 2024-03-20:
[clang-14](https://github.com/YosysHQ/yosys/actions/runs/8365812508/job/22904562564?pr=4183), it also built with clang-13, clang-16, and clang-17 just fine.  You can try `git checkout krys/config-none` and then call `make echo-cxx` (either with or without calling `make config-clang` first).  If that doesn't work for you, then there is nothing we can do as the issue is with _your_ system environment, not a problem with Yosys.

closed 2024-03-25 by mmicko