YosysHQ/yosys #4250: Synthesis fails with the latest yosys.

zapta, 2024-02-29:
### Version

Yosys 0.38+113 (git sha1 30fead6d6, clang++ 15.0.0 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

Run this command with the verilog files from the attached zip file.

yosys -p "synth_ice40 -top main -json hardware.json" leds.v main.v queue.v queue_pusher.v register.v reset_gen.v serial_tx.v sine_gen.v timing.v

[Archive 2.zip](https://github.com/YosysHQ/yosys/files/14451000/Archive.2.zip)

### Expected Behavior

Yosys completes with no error.  This command works with an older version of yosys that reports as 

Yosys 0.9+4081 (git sha1 862e84eb, x86_64-apple-darwin20.2-clang 10.0.0-4ubuntu1 -fPIC -Os)

### Actual Behavior

Yosys exists with the error:  ERROR: +/ice40/brams.txt:1: unknown top-level item `bram`.

The full log is here https://pastebin.com/jR45mB9b

povik, 2024-03-01:
Could there be something wrong with you installation? I think this would happen if you combine `ice40/brams.txt` from the older Yosys version you mention, and a newer Yosys binary.

zapta, 2024-03-01:
@povik, you are right, I had a leftover symlink.  Thanks.

closed 2024-03-01 by zapta