YosysHQ/yosys #4251: intel/common/brams_map_m9k.v widthad_* and width_* swapped

rherveille, 2024-03-02:
### Version

yosys 0.36+58

### On which OS did this happen?

Linux

### Reproduction Steps

Generated altsyncram instances have widthad_a/width_a and widthad_b/width_b parameter values swapped
Fixed by editing brams_map_m9k.v to:
.widthad_b  ( CFG_ABITS    ),
 .width_b      ( CFG_DBITS    ),
 .widthad_a ( CFG_ABITS    ),
 .width_a       ( CFG_DBITS    )

Tested and validated with experimental MAX10 flow.

### Expected Behavior

.

### Actual Behavior

.

KrystalDelusion, 2024-03-19:
Is this fixed by #4262 ?

rherveille, 2024-03-19:
It seems fixed for me. I checked the results and I can push the output verilog file (not vqm) through quartus.
You can always try the updates on a fresh git clone.

Richard

closed 2025-03-07 by povik