YosysHQ/yosys #4263: AST fails to handle implicit wires properly

ylm, 2024-03-06:
### Version

Yosys 0.38+4 (git sha1 0b835f28c, gcc 13.2.1 -fPIC -O2)

### On which OS did this happen?

Linux, Windows

### Reproduction Steps

```verilog
module gold(a, b);
	output wire [1:0] a;
	input wire [1:0] b;
	genvar i;
	for (i = 0; i < 2; i++) begin
		wire x;
		assign x = b[i];
		assign a[i] = x;
	end
endmodule

module gate(a, b);
	output wire [1:0] a;
	input wire [1:0] b;
	genvar i;
	for (i = 0; i < 2; i++) begin
		assign x = b[i];
		assign a[i] = x;
	end
endmodule
```
```tcl
read_verilog -sv genblk_wire.sv

select -assert-count 1 gate/genblk1[0].x
select -assert-count 1 gate/genblk1[1].x
select -assert-count 0 gate/genblk1[2].x

select -assert-count 1 gold/genblk1[0].x
select -assert-count 1 gold/genblk1[1].x
select -assert-count 0 gold/genblk1[2].x
```

### Expected Behavior

Both module should match.

### Actual Behavior

```
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.38+4 (git sha1 0b835f28c, gcc 13.2.1 -fPIC -O2)

-- Executing script file `genblk_wire.ys' --

1. Executing Verilog-2005 frontend: genblk_wire.sv
Parsing SystemVerilog input from `genblk_wire.sv' to AST representation.
Generating RTLIL representation for module `\gold'.
Generating RTLIL representation for module `\gate'.
genblk_wire.sv:17: Warning: Identifier `\x' is implicitly declared.
Successfully finished Verilog frontend.
ERROR: Assertion failed: selection contains 0 elements instead of the asserted 1: gate/genblk1[0].x
Selection contains:
```

povik, 2024-03-06:
So implicit wires are not created scoped to the generate blocks to which they should belong, that's the issue?

povik, 2024-03-06:
Section 4.5 of the Verilog standard:

> The implicit net declaration belongs to the scope in which the net reference appears. For example, if the implicit net is declared by a reference in a generate block, then the net is implicitly declared only in that generate block. Subsequent references to the net from outside the generate block or in another generate block within the same module either would be illegal or would create another implicit declaration of a different net (depending on whether the reference meets the above criteria). See 12.4 for information about generate blocks.

ylm, 2024-03-06:
Not only are they not scoped to the generate blocks (which isn't that problematic), they're also not properly expanded in the loop unrolling creating a single wire where there should be multiple introducing multiple drivers on that wire.

ylm, 2024-05-22:
#4235 fixes the bug, but reviewer doesn't seem to be active anymore. I don't want to be pushy or impatient, but I wouldn't want my efforts to be wasted either.