YosysHQ/yosys #4279: Top package import error

cemery123, 2024-03-13:
### Version

 Yosys 0.37+1 (git sha1 e1f4c5c9cbb, clang  -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

read Subsystem.sv

### Expected Behavior

Successfully finished Verilog frontend.

### Actual Behavior

 ERROR: syntax error, unexpected TOK_ID
It seems that Yosys can not support the package import. I have tried to read the package.sv first but it is useless 
[sampleModel14327_sub.zip](https://github.com/YosysHQ/yosys/files/14580895/sampleModel14327_sub.zip)

KrystalDelusion, 2024-03-18:
The Yosys Verilog front-end has only limited support for SystemVerilog and packages, and does not currently support the `import` keyword.  As a work-around, you can change references to `vector_of_unsigned_logic_8` to instead reference `Subsystem_pkg::vector_of_unsigned_logic_8` and drop the package import line, but that just leads to another error in the Yosys Verilog front-end.

If you want full SystemVerilog support you can [Contact YosysHQ](https://www.yosyshq.com/contact) for a [Tabby CAD Suite](https://www.yosyshq.com/tabby-cad-datasheet) Evaluation License and download link.  You might be able to try [synlig](https://github.com/chipsalliance/synlig) for an open source alternative, but I don't have any experience with that.

nakengelhardt, 2024-03-25:
Duplicate of #4006

closed 2024-03-25 by nakengelhardt