YosysHQ/yosys #4280: Performance regression in Yosys 0.39

whitequark, 2024-03-13:
### Version

0.39

### On which OS did this happen?

Linux

### Reproduction Steps

Run Glasgow Interface Explorer testcase [glasgow.applet.interface.analyzer.test.AnalyzerAppletTestCase.test_build](https://github.com/GlasgowEmbedded/glasgow/actions/runs/8262546037/job/22602091160?pr=535)

### Expected Behavior

Build passes timing.

### Actual Behavior

Build does not pass timing since Yosys 0.39.

povik, 2024-03-13:
@whitequark Do we know v0.38 is fine?

whitequark, 2024-03-13:
Yes, here is [an equivalent build using Yosys 0.38+0](https://github.com/GlasgowEmbedded/glasgow/actions/runs/8180794977/job/22369509863)

whitequark, 2024-03-13:
I actually can't see any commit that should influence this, so maybe it was just a marginal test that worked for like a dozen or two Yosys versions but started failing now?

povik, 2024-03-13:
I wonder too. I guess you can ask github to rerun the test, to see if it fails reliably?

povik, 2024-03-13:
> I guess you can ask github to rerun the test, to see if it fails reliably?

Ah, but it probably does fail reliably, because the synthesis is deterministic though chaotic. That's what you are getting at when you say it worked for a couple of Yosys versions...

whitequark, 2024-03-13:
Yes.