YosysHQ/yosys #4290: Hang on a loop when write_cxxrtl

1353369570, 2024-03-20:
### Version

Yosys 0.15 (git sha1 UNKNOWN, gcc 9.4.0-1ubuntu1~20.04 -fPIC -Os)

### On which OS did this happen?

Linux, Windows

### Reproduction Steps

my design:
module top_1  (y, clk, wire0, wire2, wire3);
  output wire [(32'h142):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire signed [(5'h14):(1'h0)] wire0;
  input wire [(5'h15):(1'h0)] wire2;
  input wire [(4'hb):(1'h0)] wire3;
  reg signed [(5'h14):(1'h0)] reg4 = (1'h0);
  reg [(5'h14):(1'h0)] reg5 = (1'h0);
  reg signed [(5'h14):(1'h0)] reg6 = (1'h0);
  assign y = y;
  always
    @(posedge clk) begin
      if (wire3)
        begin
          reg3 <= reg3;
          reg4 <= $signed(reg4[(2'h2):(1'h0)]);
        end
      else
        begin
          reg4 <= wire2;
          reg5 <= (~|($unsigned($signed({wire3, reg4})) ?
              $unsigned((!{wire0, wire0})) : (reg4[(1'h1):(1'h1)] ?
                  $unsigned($signed(reg4)) : ($signed(wire0) & (!(8'hbb))))));
          reg6 <= $signed(reg4);
        end
    end
endmodule

when yosys-p "read_verilog rtl.v；hierarchy -top top; write_cxxrtl  rtl_cxxrtl.v"

### Expected Behavior

The process will either Execute the xxx pass or report  Error.

### Actual Behavior

The write_cxxrtl process hangs indefinitely following:
  Module `top_1' contains feedback arcs through wires:
       y

whitequark, 2024-03-20:
> Yosys 0.15 (git sha1 UNKNOWN, gcc 9.4.0-1ubuntu1~20.04 -fPIC -Os)

This Yosys version is several years old. Unless you can reproduce the report with a newer version I will close it.

1353369570, 2024-04-01:
When use "osys 0.39+149 (git sha1 b9d3bffda, g++ 9.4.0-1ubuntu1~20.04 -fPIC -Os)",the The write_cxxrtl process hangs indefinitely following:
4.3.12. Executing OPT_EXPR pass (perform const folding).
Optimizing module top_1.
<suppressed ~1 debug messages>

Module `top_1' contains feedback arcs through wires:
  y

[image]

AHarald, 2024-04-18:
Hi whitequark, 
I encounter the same problem with 
Yosys 0.40+25 (git sha1 171577f90, g++ 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)

The program write_cxxrtl hangs in an endless loop when executing: 
Module `sigmadelta' contains feedback arcs through wires:
  adc.fir.ram.n980_q
  adc.fir.ram.n982_q

Yosys runs on mint linux.
If you need the verilog source for tsting, please öet me know.