YosysHQ/yosys #4291: ERROR: Visited AIG node more than once; this could be a combinatorial loop that has not been broken

DmitryZlobec, 2024-03-21:
### Version

ERROR: Visited AIG node more than once; this could be a combinatorial loop that has not been broken

### On which OS did this happen?

Linux

### Reproduction Steps

 yosys  top.sv  tm1638_board.sv  -p "synth_ecp5 -json top.json -top top"
[err.gz](https://github.com/YosysHQ/yosys/files/14703575/err.gz)
[tm1638_board.sv.gz](https://github.com/YosysHQ/yosys/files/14703576/tm1638_board.sv.gz)
[top.sv.gz](https://github.com/YosysHQ/yosys/files/14703577/top.sv.gz)

### Expected Behavior

synthesis completes without  using -noabc9

### Actual Behavior

3.43.6. Executing SCC pass (detecting logic loops).
Found 0 SCCs in module top.
Found 0 SCCs.
....
3.43.19.3. Executing XAIGER backend.
ERROR: Visited AIG node more than once; this could be a combinatorial loop that has not been broken

Ravenslofty, 2024-03-21:
Minimised testcase:
```verilog
module top(inout [11:0] GPIO, input _0_, input _1_, input _2_);
  reg \GPIO_reg[1] ;
  always @(posedge _0_, posedge _1_)
    if (_1_) \GPIO_reg[1]  <= 1'b1;
    else \GPIO_reg[1]  <= _2_;
  assign GPIO[1] = \GPIO_reg[1] ;
endmodule
```

Ravenslofty, 2024-03-21:
I think this might be a variant of #4168.

terracide303, 2026-08-19:
Small correction to my comment above: I checked my install and the version I was on was actually 0.68+48, not 0.68+58. Sorry about that.

I have now updated to the 2026-08-19 nightly, Yosys 0.68+86 (9eb62484d), on macOS arm64, and the minimised testcase from this thread still fails:

```
2.46.16.3. Executing XAIGER backend.
ERROR: Visited AIG node more than once; this could be a combinatorial loop that has not been broken
```

I get the same error on 0.68+48 and on 0.68+86, so it does not look like a recent regression.

One thing I noticed while testing: synth_ecp5 seems to run abc9 by default now. If I run plain synth_ecp5 with no flags it still goes through ABC9_OPS and XAIGER and fails the same way. Only synth_ecp5 -noabc9 gets through, in about 0.3 seconds. So you do not have to ask for abc9 to hit this any more.

Ravenslofty, 2026-08-19:
Well, [soon](https://github.com/YosysHQ/yosys/pull/6103) `-noabc9` will still give you ABC9.

Ravenslofty, 2026-08-19:
It's true that the error message is misleading; I think this is actually to do with the `inout` port in the minimised testcase. Bidirectional port handling is *hard* in ABC9 because the XAIGER format does not natively support bidirectional ports.

Ravenslofty, 2026-09-03:
Worth mentioning that with #6174 the minimised testcase and original testcase both synthesise, although `check` spits out a lot of warnings. So I still can't really recommend using `inout`s in logic.