YosysHQ/yosys #4308: yosys fails with 'ERROR: init_share_dirname: unable to determine share/ directory!' on macos

conversy, 2024-04-01:
### Version

Yosys 0.39+149 (git sha1 b9d3bffda, aarch64-apple-darwin21.4-clang++ 14.0.0-1ubuntu1.1 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

calling `yosys` from `oss-cad-suite` like this: `yosys -q -p "synth_gowin -json build.json -top top" build.v`

### Expected Behavior

it should not fail.

### Actual Behavior

it fails with: `ERROR: init_share_dirname: unable to determine share/ directory!`

nakengelhardt, 2024-04-02:
I can't reproduce this locally with either the oss-cad-suite or the local build. Did you remove the quarantine attribute on the archive before unpacking? (`xattr -d com.apple.quarantine oss-cad-suite-darwin-arm64-20240402.tgz`)
Did you perhaps place the suite in one of the directories that require granting special permissions (e.g. Documents)? Are there any spaces in its path?

conversy, 2024-04-02:
Thanks for helping.

Here is what I do to chase the problem:
`source ../../../oss-cad-suite-2024-04-01/environment`
=> I get the error when using yosys
`source ../../../oss-cad-suite-2024-03-02/environment`
=> no error

So I guess it's not a problem of quarantine or space in path... I did not try to bisect the oss-cad-suite distribution, but I can if it helps. I went back to the beginning of March because the last commit on `yosys.cc` (which is about `proc_self_dirname` incidently) is dated around this time...: https://github.com/YosysHQ/yosys/commits/main/kernel/yosys.cc

conversy, 2024-04-03:
my bad.
I erroneously thought that `yosys` was the  culprit, but it's actually `nextpnr`.
They happen to have the same error message in the source code, and as I found it in `yosys`' code, I did not check for its presence in another source code :-/
Sorry for the inconvenience, I'll look into `nextpnr` then...

closed 2024-04-03 by conversy