YosysHQ/yosys #4318: Parameters in other packages

pentin-as, 2024-04-04:
### Version

Yosys 0.39 (git sha1 00338082b, clang++ 14.0.6 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

package my_const;
  localparam int MY_CONST = 5;
endpackage

package test1;
  function int test1();
    test1 = my_const::MY_CONST; // <- ok
  endfunction
endpackage

package test2;
  localparam int MY_CONST = my_const::MY_CONST; // <- ERROR: Identifier `\my_const::MY_CONST' is implicitly declared outside of a module
  function int test2();
    test2 = MY_CONST;
  endfunction
endpackage

### Expected Behavior

Both packages can be compiled

### Actual Behavior

package test2 throws an error:
ERROR: Identifier `\my_const::MY_CONST' is implicitly declared outside of a module

KrystalDelusion, 2024-04-04:
SystemVerilog packages only have very limited support in the verilog front end, so I suspect this is related.

zachjs, 2024-04-04:
You could try https://github.com/zachjs/sv2v, which has full support for SystemVerilog packages.