YosysHQ/yosys #4336: Inconsistency in Verilog Synthesis: Yosys Successfully Synthesizes Code That Fails in Vivado and Quartus Due to Syntax Errors

1353369570, 2024-04-14:
### Version

Yosys 0.39+149 

### On which OS did this happen?

Linux

### Reproduction Steps

my test

module top  (y, clk, wire0, wire2, wire3, wire4);
  output wire [(32'ha2):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire [(5'h15):(1'h0)] wire0;
  input wire signed [(5'h12):(1'h0)] wire2;
  input wire signed [(5'h10):(1'h0)] wire3;
  input wire [(3'h5):(1'h0)] wire4;
  wire signed [(5'h11):(1'h0)] wire194;
  assign y = {wire194};
  module6 modinst195 (wire194, clk, wire0, wire2, wire3, (1'h0), wire4);
endmodule
module module6  (y, clk, wire7, wire8, wire9, wire10, wire11);
  output wire [(32'h2b7):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire [(5'h15):(1'h0)] wire7;
  input wire signed [(5'h12):(1'h0)] wire8;
  input wire signed [(3'h7):(1'h0)] wire9;
  input wire signed [(5'h10):(1'h0)] wire10;
  input wire [(3'h5):(1'h0)] wire11;
  reg [(2'h2):(1'h0)] reg16 = (1'h0);
  reg signed [(5'h15):(1'h0)] reg17 = (1'h0);
  reg signed [(5'h11):(1'h0)] reg18 = (1'h0);
  reg [(4'ha):(1'h0)] reg19 = (1'h0);
  reg signed [(5'h14):(1'h0)] reg20 = (1'h0);
  reg [(5'h15):(1'h0)] reg21 = (1'h0);
  reg signed [(4'h9):(1'h0)] reg22 = (1'h0);
  reg [(4'hf):(1'h0)] reg23 = (1'h0);
  reg signed [(4'hb):(1'h0)] reg24 = (1'h0);
  assign y = {reg16, reg17, reg18, reg19, reg20, reg21, reg22, reg23, reg24};
  always
    @(posedge clk) begin
      if (($unsigned(((1'h0) <= $signed(wire8[(5'h10):(4'h8)]))) ?
          reg17[(2'h2):(1'h0)] : (!$unsigned(reg16[(2'h2):(1'h0)]))))
        begin
          reg16 <= (^~wire11);
          reg17 <= $unsigned($signed((wire10[(3'h5):(1'h0)] ?
              reg20 : wire10[(4'hf):(1'h1)])));
          reg18 <= ({{$unsigned((wire7 | reg16)),
                  wire10[(4'h8):(4'h8)]}} ^~ (8'hb2));
          reg19 <= {((((|wire10) != {wire7}) ?
                  {reg19[(3'h6):(2'h3)],
                      (&(1'h0))} : (~$signed((8'h9c)))) > $signed($unsigned(wire10))),
              {$unsigned((^((8'hbb) & wire7))),
                  $signed($unsigned(((1'h0) ? (1'h0) : (1'h0))))}};
        end
      else
        begin
          reg16 <= {reg20,
              $signed($signed(($signed((7'h43)) ?
                  ((8'hbf) ? reg18 : reg17) : $unsigned(reg16))))};
          reg17 <= ($signed(reg18) ?
              $signed($signed(((reg18 ? wire10 : reg16) ?
                  $signed(wire11) : $unsigned((1'h0))))) : (~^(^reg17)));
          reg18 <= $unsigned((^reg20[(4'hd):(4'h8)]));
        end
      if ((wire11[(1'h1):(1'h0)] ? (1'h0) : reg16[(2'h2):(1'h0)]))
        begin
          reg20 <= ($signed($unsigned((1'h0))) ?
              {((~^reg18[(4'ha):(3'h5)]) >> wire10[(4'ha):(3'h4)]),
                  ((reg17[(1'h0):(1'h0)] + (-wire11)) == ((reg20 ?
                      wire11 : wire8) || ((8'ha9) + wire8)))} : wire7);
          reg21 <= (reg19 || (1'h0));
          reg22 <= (!$signed(wire10));
         // 出现异常的代码片段
          reg16 <=  ( reg16 ** reg18 ) ** ( !reg17 ) ;
          reg24 <=  !( reg19 - wire7 ) ;
          reg22 <=  ( reg17 - wire9 ) == ( reg16 ** reg18 ) ;
          reg21 <=  ( reg16 ** reg18 ) == ( wire11 - wire8 ) ;
          reg19 <=  ( reg16 ** reg18 ) - ( reg17 - wire9 ) ;
          reg24 <=  ( reg20 - wire9 ) - ( reg17 - wire9 ) ;
          reg22 <=  ( reg20 - wire9 ) - ( reg16 ** reg18 ) ;
          reg24 <=  ( wire11 - wire8 ) - ( reg16 ** reg18 ) ;
          reg23 <= (-$unsigned($signed({wire10})));
          reg24 <= {reg21[(2'h3):(2'h2)]};
        end
      else
        begin
          reg20 <= $signed({reg20[(1'h0):(1'h0)]});
          reg21 <= $unsigned({wire9, wire7[(2'h3):(1'h1)]});
          reg22 <= (8'hba);
          reg23 <= ($signed(reg19) ?
              $signed($signed((~&(~®16)))) : (-$unsigned($unsigned($unsigned((8'hab))))));
          reg24 <= (~reg20);
        end
    end
endmodule

### Expected Behavior

Like Vivado and Quartus ,fails   Due to Syntax Errors"
[image]

### Actual Behavior

Yosys Successfully Synthesizes Code

1353369570, 2024-04-20:
I think this is already the most minimized design, and this is how it performs in Quartus:
[image]

nakengelhardt, 2024-04-20:
No, this would be a minimized design:
```
module top(input [7:0] a, input [2:0] b, output [7:0] o);
    assign o = a ** b;
endmodule
```

Closing in favor of #4348.

closed 2024-04-20 by nakengelhardt