YosysHQ/yosys #4349: "ERROR: Assert `count_id(wire->name) == 0' failed in kernel/rtlil.cc:2143" when using synth_{ice40,ecp5} on simple design

cr1901, 2024-04-20:
### Version

Yosys 0.40+25 (git sha1 171577f90, sccache x86_64-w64-mingw32-g++ 13.2.0 -Os)

### On which OS did this happen?

Windows

### Reproduction Steps

1. Save this file as `bad.rtlil`:
    ```
    # Generated by Yosys 0.40+25 (git sha1 171577f90, sccache x86_64-w64-mingw32-g++ 13.2.0 -Os)
    autoidx 95
    attribute \src "C:/msys64/home/William/Projects/FPGA/amaranth/smolarith/src/smolarith/base10.py:347"
    attribute \generator "Amaranth"
    attribute \hdlname "top"
    attribute \top 1
    module \top
      attribute \src "b2tob1000.v:143.3-152.6"
      wire width 11 $0\$15[10:0]$34
      attribute \src "b2tob1000.v:173.3-189.6"
      wire width 20 $0\$18[19:0]$46
      attribute \src "b2tob1000.v:91.3-92.23"
      wire width 11 $0\d0_guess[10:0]
      attribute \src "b2tob1000.v:100.3-101.28"
      wire width 20 $0\outp__payload[19:0]
      attribute \src "b2tob1000.v:143.3-152.6"
      wire width 11 $1\$15[10:0]$35
      attribute \src "b2tob1000.v:173.3-189.6"
      wire width 20 $1\$18[19:0]$47
      attribute \src "b2tob1000.v:143.3-152.6"
      wire width 11 $2\$15[10:0]$36
      attribute \src "b2tob1000.v:173.3-189.6"
      wire width 20 $2\$18[19:0]$48
      attribute \src "b2tob1000.v:173.3-189.6"
      wire width 20 $3\$18[19:0]$49
      attribute \src "b2tob1000.v:76.17-76.128"
      wire width 11 $add$b2tob1000.v:76$7_Y
      attribute $bugpoint 1
      wire $auto$bugpoint.cc:258:simplify_something$59
      attribute $bugpoint 1
      wire $auto$bugpoint.cc:258:simplify_something$61
      attribute $bugpoint 1
      wire width 10 input 3 $auto$bugpoint.cc:258:simplify_something$63
      attribute $bugpoint 1
      wire width 10 $auto$bugpoint.cc:258:simplify_something$64
      wire $delete_wire$66
      wire $delete_wire$67
      wire $delete_wire$68
      wire $delete_wire$69
      wire $delete_wire$70
      wire $delete_wire$73
      wire $delete_wire$74
      wire $delete_wire$75
      wire $delete_wire$76
      wire $delete_wire$77
      wire $delete_wire$78
      wire $delete_wire$79
      attribute \src "b2tob1000.v:14.14-14.18"
      wire width 11 \$15
      attribute \src "b2tob1000.v:17.14-17.18"
      wire width 20 \$18
      attribute \src "b2tob1000.v:23.15-23.18"
      wire width 11 \$7
      attribute \src "C:/msys64/home/William/Projects/FPGA/amaranth/smolarith/.venv/lib/python3.11/site-packages/amaranth/hdl/_ir.py:270"
      wire input 1 \clk
      attribute \src "C:/msys64/home/William/Projects/FPGA/amaranth/smolarith/src/smolarith/base10.py:351"
      wire width 11 \d0_guess
      attribute \src "C:/msys64/home/William/Projects/FPGA/amaranth/smolarith/src/smolarith/base10.py:333"
      wire width 20 output 2 \outp__payload
      attribute \src "C:/msys64/home/William/Projects/FPGA/amaranth/smolarith/src/smolarith/base10.py:333"
      wire width 10 \outp__payload[0]
      attribute \src "C:/msys64/home/William/Projects/FPGA/amaranth/smolarith/src/smolarith/base10.py:323"
      cell $add $add$b2tob1000.v:76$7
        parameter \A_SIGNED 0
        parameter \A_WIDTH 10
        parameter \B_SIGNED 0
        parameter \B_WIDTH 10
        parameter \Y_WIDTH 11
        connect \A $auto$bugpoint.cc:258:simplify_something$63
        connect \B $auto$bugpoint.cc:258:simplify_something$64
        connect \Y $add$b2tob1000.v:76$7_Y
      end
      attribute \src "C:/msys64/home/William/Projects/FPGA/amaranth/smolarith/src/smolarith/base10.py:333"
      process $proc$b2tob1000.v:100$18
        assign $0\outp__payload[19:0] \$18
        sync posedge \clk
          update \outp__payload $0\outp__payload[19:0]
      end
      attribute \src "b2tob1000.v:143.3-152.6"
      process $proc$b2tob1000.v:143$33
        assign $0\$15[10:0]$34 $2\$15[10:0]$36
        attribute \src "b2tob1000.v:144.5-144.65"
        switch $delete_wire$74
          case 1'1
          case 
        end
        attribute \src "b2tob1000.v:146.5-148.8"
        switch $delete_wire$70
          case 1'1
            assign $1\$15[10:0]$35 \$7
          case 
        end
        attribute \src "b2tob1000.v:149.5-151.8"
        switch $delete_wire$67
          case 1'1
          case 
            assign $2\$15[10:0]$36 $1\$15[10:0]$35
        end
        sync always
          update \$15 $0\$15[10:0]$34
      end
      attribute \src "b2tob1000.v:173.3-189.6"
      process $proc$b2tob1000.v:173$45
        assign $0\$18[19:0]$46 $3\$18[19:0]$49
        attribute \src "b2tob1000.v:174.5-174.65"
        switch $delete_wire$73
          case 1'1
          case 
        end
        attribute \src "b2tob1000.v:176.5-185.8"
        switch $delete_wire$69
          case 1'1
            assign $1\$18[19:0]$47 $2\$18[19:0]$48
            attribute \full_case 1
            attribute \src "b2tob1000.v:178.7-184.10"
            switch $delete_wire$76
              case 1'1
              case 
                assign $2\$18[19:0]$48 [9:0] \d0_guess [9:0]
            end
          case 
        end
        attribute \src "b2tob1000.v:186.5-188.8"
        switch $delete_wire$66
          case 1'1
          case 
            assign $3\$18[19:0]$49 $1\$18[19:0]$47
        end
        sync always
          update \$18 $0\$18[19:0]$46
      end
      attribute \src "C:/msys64/home/William/Projects/FPGA/amaranth/smolarith/src/smolarith/base10.py:351"
      process $proc$b2tob1000.v:91$15
        assign $0\d0_guess[10:0] \$15
        sync posedge \clk
          update \d0_guess $0\d0_guess[10:0]
      end
      connect { } { }
      connect { } { }
      connect { } { }
      connect { } { }
      connect { } { }
      connect { } { }
      connect \$7 $add$b2tob1000.v:76$7_Y
      connect { } { }
      connect { } { }
      connect { } { }
      connect { } { }
      connect { } { }
      connect \outp__payload[0] \outp__payload [9:0]
      connect { } { }
      connect { } { }
      connect { } { }
    end
    ```

2. Run `yosys -f rtlil -p synth_ice40 bad.rtlil`

### Expected Behavior

_`bad.rtlil` is a testcase generated by `yosys -p 'hierarchy; bugpoint -command "synth_ice40"; write_rtlil bad.rtlil' b2tob1000.v`_. I expect `yosys` to successfully synthesize the `b2tob1000.v` and get to the `stats` pass without crashing. I provide `b2tob1000.v` [here](http://gopher.wdj-consulting.com:70/paste/e48f875b-04ea-47dc-952d-52a18a4121ea.txt).

### Actual Behavior

When I run _either_ `bad.rtlil` (see Reproduction Steps) or `b2tob1000.v` (`yosys -p synth_ice40 b2tob1000.v`), I instead get a crash similar to the following:

```
2.41.16.6. Executing AIGER frontend.
<suppressed ~154 debug messages>
ERROR: Assert `count_id(wire->name) == 0' failed in kernel/rtlil.cc:2143.
```

This is independent of OS.

## Additional Context

For the `b2tob1000.v` case, I have locally bisected the bug to commit 8e470add4d1.

Ravenslofty, 2024-05-08:
(this reply is mostly me thinking out loud, in the hopes that if I get distracted I can come back to this and not have to redo too much work)

So, the quirky bit of the testcase here is that you have two wires: `\outp__payload` and `\outp__payload[0]`.

Inside the AIGER parser there is a little section of code which assumes that if it has encountered a wire before, it is the least significant bit of a wide wire, and renames it accordingly:

https://github.com/YosysHQ/yosys/blob/77dff5a293b1baa2e4d10ef80558bf49395eab48/frontends/aiger/aigerparse.cc#L923-L925

except here it renames `\outp__payload` to `\outp__payload[0]`, which is a wire that already exists, and so Yosys asserts.

Ravenslofty, 2024-05-08:
Okay, the "proper" fix to this would be to detect this situation and rename wires and box ports appropriately (and perhaps this issue exists in other parts of the codebase).

_Or_ I can insert a strategic `opt_clean -purge` and the problem goes away in both testcases.

cr1901, 2024-05-22:
Temporary workaround: Something like `yosys -p "synth_ice40 -run :coarse; opt_clean -purge; synth_ice40 -run coarse:" input.v`

ld-cd, 2026-09-15:
After quite a bit of spelunking (see https://libera.catirclogs.org/glasgow/2026-09-15#1789508854-1789513090), it seems that this is solved in `main` by switching to the aiger2 backend, i was unable to track down the original unsoundness, and i have surrendered to just not generating rtlil that looks like that ™️ until there is another yosys release.

for what it is worth https://paste.sr.ht/~ld-cd/1dd92c5ab2bd030267949ded1a0fcc672aa4ea71 is a ~minified repro of the issue

Ravenslofty, 2026-09-16:
Do you mind if I adopt that minified repro as a testcase for xaiger2?

ld-cd, 2026-09-16:
Nope go ahead