YosysHQ/yosys #4357: opt: no "-purge" option but public names removed

YikeZhou, 2024-04-25:
### Version

Yosys 0.40+25 (git sha1 171577f90, clang++ 14.0.6 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Before opening this issue, I found some really useful information in #2165.
Though, it was 4 years ago.
Therefore, I'm not sure if the following behavior of Yosys is expected but undocumented, or a bug.

Please consider this simple example:
```verilog
module top(input clk, input signed [10:0] i);
  reg [2:0] r;
  always @(posedge clk)
    r <= i;
endmodule
```

Command:
```
yosys -p "read_verilog input.v; proc; opt; write_rtlil"
```

### Expected Behavior

Having read the sections for `opt` and `opt_clean` in Yosys manual, I expected `r` to be remained after `opt` was called.

### Actual Behavior

When running the command above, I noticed that:
```
3.7. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \top..
Removed 1 unused cells and 2 unused wires.
<suppressed ~3 debug messages>
```

It seems that `opt_clean` removed `r` and it is missing in the RTLIL output.
```
# Generated by Yosys 0.40+25 (git sha1 171577f90, clang++ 14.0.6 -fPIC -Os)
autoidx 3
attribute \cells_not_processed 1
attribute \src "input.v:1.1-5.10"
module \top
  attribute \src "input.v:1.18-1.21"
  wire input 1 \clk
  attribute \src "input.v:1.43-1.44"
  wire width 11 input 2 signed \i
end
```

whitequark, 2024-04-25:
Keeping these kinds of unused registers is something that's potentially important for Amaranth and CXXRTL simulations, so I think we need at least a clear way to disable the behavior.

nakengelhardt, 2024-04-25:
> Keeping these kinds of unused registers is something that's potentially important for Amaranth and CXXRTL simulations, so I think we need at least a clear way to disable the behavior.

This was discussed in the context of sby a few years ago and `setattr -set keep 1 w:\*` is currently the way to disable that behavior...