YosysHQ/yosys #4358: Tests fails on Debian GNU/Linux on ppc64

spth, 2024-04-25:
### Version

Yosys 0.40 (git sha1 a1bb0255d65, g++ 13.2.0-19 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Download yosys 0.40 tarball, build on Debian GNU/Linux on ppc64 using "make", then run the tests using "LANG=C make tests".

### Expected Behavior

All tests pass (as they do for me on Debian GNU/Linux on amd64 and Debian GNU/Linux on arm64).

### Actual Behavior

~~~~
…
Passed iopadmap.ys
Passed kogge-stone.ys
Passed pmux2mux.ys
ERROR: Called with -verify and proof did fail!
make[1]: *** [run-test.mk:216: shiftx2mux.ys] Error 1
make[1]: Leaving directory '/home/philipp/yosys-yosys-0.40/tests/techmap'
make: *** [Makefile:865: test] Error 2
philipp@nemesis:~/yosys-yosys-0.40$
~~~~

KrystalDelusion, 2024-04-25:
Out of curiosity, what is the purpose of the `LANG=C` env var?

spth, 2024-04-25:
My default is

~~~~
philipp@notebook6:~$ echo $LANG
de_DE.UTF-8
~~~~

So without the LANG=C, I get German error messages, which might be confusing for others reading this ticket.

spth, 2024-04-25:
I see the problem both with and without LANG=C. I have no reason to assume that LANG=C affects the problem happening. But without LANG=C, the error message will be in the user's local language.

wanda-phi, 2024-04-26:
duplicate of #2645

whitequark, 2024-04-26:
Duplicate of #2645?

closed 2024-04-26 by KrystalDelusion