YosysHQ/yosys #4371: Inout port not working as expected

alchitry, 2024-05-03:
### Version

Yosys 0.40+45 (git sha1 dd2195543, g++ 13.2.1 -march=x86-64 -mtune=generic -O2 -fstack-protector-strong -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

I'm currently working on the new Lucid to Verilog translator for Alchitry Labs and come across a weird behaviour when working with `inout` ports and the array replication operator.

``` verilog
module MI_alchitryTop (
    input P_clk,
    output reg [7:0] P_led,
    inout [4:0] P_button
  );
  
  reg [3:0] D_flip_d, D_flip_q = 4'h0;
  
  reg [4:0] IO_P_button;
  assign P_button = IO_P_button;
  
  always @* begin
    D_flip_d = D_flip_q + 1'h1;
    IO_P_button = {3'h5{D_flip_q ? 1'bz : 1'h0}};
    P_led = P_button;
  end
  
  always @(posedge (P_clk)) begin
    D_flip_q <= D_flip_d;
  end
endmodule
```

With the replication syntax, all 5 LEDs light up when the first button is pressed.

If I change the line `IO_P_button = {3'h5{D_flip_q ? 1'bz : 1'h0}};` to `IO_P_button = D_flip_q ? 5'bzzzzz : 5'h0;` it works as expected with each LED tied to its own button.

I'm targeting the iCE40HX8K-CB132

### Expected Behavior

When a button is pressed the corresponding LED lights up.

### Actual Behavior

When the first (LSB) button is pressed, it lights up all the LEDs.

KrystalDelusion, 2024-05-04:
Is that supposed to be `IO_P_button = {3'h5, {D_flip_q ? 1'bz : 1'h0}};` with a comma?

alchitry, 2024-05-06:
No, it is supposed to be the replication operator `{x{array}}` where `array` gets duplicated `x` times. See about halfway down [this page](https://asic-world.com/verilog/operators2.html)

KrystalDelusion, 2024-05-06:
Ah I see, I didn't expect the 3'h5 to work in that context.  I suspect this is the same underlying issue as #4370 with `read_verilog` prematurely optimising all of the `z` values to a single comparison and repeating the output for each.

whitequark, 2024-05-07:
@alchitry From my experience in Amaranth I would recommend not letting Yosys deal with `inout`s  and instead instantiate vendor primitives directly, since that's guaranteed to give you a repeatable result.

alchitry, 2024-05-07:
If this was just for a personal project, I would likely do just that. However, this is from the Lucid -> Verilog translator and making that generate vendor primitives would be the same as updating Yosys to do the same.

For now, I understand the limitation and I'll document it in our tutorials.

whitequark, 2024-05-07:
> making that generate vendor primitives would be the same as updating Yosys to do the same

It would be less effort though since you no longer need to care about the batshit Verilog semantics.

closed 2024-05-08 by KrystalDelusion