YosysHQ/yosys #4375: Unexpected Result from `synth_gowin` Pass

Seyviour, 2024-05-05:
### Version

Yosys 0.40+33 (git sha1 34d9a7451, clang++ 14.0.0-1ubuntu1.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

I observed a potential error where 'ORing' an undriven net with an input net results in a `HIGH` output, after executing `synth_gowin`. The following example reproduces the potential error -- in this case, `buggy_a` is set `HIGH` when `bug_sub` is instantiated in `bug_top`.

```
module bug_sub(
    input wire clk,
    input wire buggy_a,
    output wire x
); 
reg [24:0] count; 
always @(posedge clk)
    if (buggy_a)
        count <= count + 1'b1; 
    
assign x = count[24]; 
endmodule

module bug_top(
    input wire clk,
    input wire a, 
    input wire b,
    output wire x
);`
    wire undriven_wire;
    // assign undriven_wire = 1'b0; 
    bug_sub this_sub (
        .clk(clk),
        .buggy_a(a | undriven_wire),
        .x(x)
    );

endmodule
```
The following line from the log (`trace` on) showss that  `buggy_a` is indeed driven high:
```
#TRACE# Connection in module bug_top: \this_sub.buggy_a = 1'1
```

The potential error seems to disappear if I do one of the following:
- explicitly assign `undriven_wire` to `1'b0` (as in the commented out line)
- execute `setundef` earlier than `synth_gowin` executes it (in a custom synthesis script that otherwise mirrors `synth_gowin` 

### Expected Behavior

`this_sub.a` should take on the the value of `a`, as `undriven_wire` is expected to be low (`synth_gowin` executes `setundef -zero`) .

### Actual Behavior

`this_sub.a` is driven `HIGH`.

whitequark, 2024-05-06:
I'm fairly sure this is not a bug, and the optimizer is allowed to assume that undriven nets take any value that it finds convenient.

Seyviour, 2024-05-06:
I would make the argument that it is a bug from the POV of intention. If I'm not misinterpreting what setundef does, then I believe `synth_gowin` 'intends' to set undriven nets to 0.

In addition to that, the undriven net itself is not really being driven `HIGH`, only the result of the expression it is a part of. When I try to assign the undriven net to an output, that output is grounded. To demonstrate this, I added an additional output, `undriven_wire_out` to `bug_top` and assigned `undriven_wire` to it. 

These are the final assignments to the nets per the log: 

```
#TRACE# Connection in module bug_top: \this_sub.buggy_a = 1'1
#TRACE# Connection in module bug_top: \undriven_wire = 1'x
#TRACE# Connection in module bug_top: \undriven_wire_out = 1'x
```

And from the json output, I can see that: 

```
 "undriven_wire_GND_G": {
          "hide_name": 0,
          "type": "GND",
          "parameters": {
          },
          "attributes": {
          },
          "port_directions": {
            "G": "output"
          },
          "connections": {
            "G": [ 19 ]
          }
        },
        "undriven_wire_out_OBUF_O": {
          "hide_name": 0,
          "type": "OBUF",
          "parameters": {
          },
          "attributes": {
            "keep": "00000000000000000000000000000001"
          },
          "port_directions": {
            "I": "input",
            "O": "output"
          },
          "connections": {
            "I": [ 19 ],
            "O": [ 6 ]
          }
        },
```

So it appears I'm ending up with a case where:

```
buggy_a = (a | undriven_wire) = 1'1 (undriven_wire is assumed to be 1'1)
```
but
```
undriven_wire_out = undriven_wire = 1'0 (per the json output above, `undriven wire` is grounded) 
```

This is consistent with the results I got from running this on an actual device, and very similar to the original challenge I encountered. Even if the optimizer makes the choice to set `undriven_wire` HIGH, doesn't it have to be consistent with that decision?

whitequark, 2024-05-07:
> Even if the optimizer makes the choice to set `undriven_wire` HIGH, doesn't it have to be consistent with that decision?

Nope, each time `'x` is replicated it can take a different value, similar to e.g. `undef` in LLVM. This is a deliberate part of its semantics.

Seyviour, 2024-05-07:
That's fair enough. Even if it is certainly jarring to encounter. 

Maybe it would be nice to have an option to tell synth scripts like `synth_gowin` to set all undriven nets to 0 prior to optimization.

Thank you very much for your time :pray: 

I'll go ahead to close the issue.

closed 2024-05-08 by Seyviour