YosysHQ/yosys #4383: Build error: `make: *** [Makefile:810: abc/abc] Error 2`

dopamane, 2024-05-09:
### Version

Yosys 0.37+1 (git sha1 37a6c9a09, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

```
git pull
make -j8
```

### Expected Behavior

Build without error.

### Actual Behavior

```
...
[ 99%] Building techlibs/quicklogic/ql_dsp_macc.o
[ 99%] Building abc/abc
make: the '-f' option requires a non-empty string argument
Usage: make [options] [target] ...
Options:
  -b, -m                      Ignored for compatibility.
  -B, --always-make           Unconditionally make all targets.
  -C DIRECTORY, --directory=DIRECTORY
                              Change to DIRECTORY before doing anything.
  -d                          Print lots of debugging information.
  --debug[=FLAGS]             Print various types of debugging information.
  -e, --environment-overrides
                              Environment variables override makefiles.
  --eval=STRING               Evaluate STRING as a makefile statement.
  -f FILE, --file=FILE, --makefile=FILE
                              Read FILE as a makefile.
  -h, --help                  Print this message and exit.
  -i, --ignore-errors         Ignore errors from recipes.
  -I DIRECTORY, --include-dir=DIRECTORY
                              Search DIRECTORY for included makefiles.
  -j [N], --jobs[=N]          Allow N jobs at once; infinite jobs with no arg.
  -k, --keep-going            Keep going when some targets can't be made.
  -l [N], --load-average[=N], --max-load[=N]
                              Don't start multiple jobs unless load is below N.
  -L, --check-symlink-times   Use the latest mtime between symlinks and target.
  -n, --just-print, --dry-run, --recon
                              Don't actually run any recipe; just print them.
  -o FILE, --old-file=FILE, --assume-old=FILE
                              Consider FILE to be very old and don't remake it.
  -O[TYPE], --output-sync[=TYPE]
                              Synchronize output of parallel jobs by TYPE.
  -p, --print-data-base       Print make's internal database.
  -q, --question              Run no recipe; exit status says if up to date.
  -r, --no-builtin-rules      Disable the built-in implicit rules.
  -R, --no-builtin-variables  Disable the built-in variable settings.
  -s, --silent, --quiet       Don't echo recipes.
  -S, --no-keep-going, --stop
                              Turns off -k.
  -t, --touch                 Touch targets instead of remaking them.
  --trace                     Print tracing information.
  -v, --version               Print the version number of make and exit.
  -w, --print-directory       Print the current directory.
  --no-print-directory        Turn off -w, even if it was turned on implicitly.
  -W FILE, --what-if=FILE, --new-file=FILE, --assume-new=FILE
                              Consider FILE to be infinitely new.
  --warn-undefined-variables  Warn when an undefined variable is referenced.

This program built for x86_64-pc-linux-gnu
Report bugs to <bug-make@gnu.org>
make: *** [Makefile:810: abc/abc] Error 2
make: *** Waiting for unfinished jobs....
```

whitequark, 2024-05-09:
Caused by https://github.com/YosysHQ/yosys/pull/4243.

whitequark, 2024-05-09:
@RCoeurjoly The cause of this issue is an incomplete check:

```
	@if [ ! -d "$(YOSYS_SRC)/abc" ]; then \
	    echo "Error: The 'abc' directory does not exist."; \
			echo "Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule."; \
	    exit 1; \
```

This isn't actually enough to check that the submodule is uninitialized. Also, you should not be assuming that abc comes from a tarball for the sole reason of "every other check has failed".

dopamane, 2024-05-09:
⚡ Thanks for the heads up, `git submodule update --init` after clone does the trick, I can open a PR which adds this additional step to the README "Building from source" section.

whitequark, 2024-05-09:
The check needs to be fixed in the Makefile as well, since no one with existing checkouts is going to diff the README after pulling.

closed 2024-05-09 by dopamane