YosysHQ/yosys #4397: The negation operation on an empty string results in an exception.

WeneneW, 2024-05-14:
### Version

Yosys 0.39+165

### On which OS did this happen?

Linux

### Reproduction Steps

Hello, I have been studying string designs in Verilog recently.
Here is the original Verilog design.
```
module top (y, clk);
  output wire [4:0] y;
  input wire clk; 

  wire [4:0] wire1; 

  assign wire1 = $signed((~""));
  assign y = wire1;
endmodule

```

The original Verilog design appears to treat empty characters as zero during testing with the testbench, resulting in an output of **1**. However, after synthesis with Yosys, the output is instead **0**. The synthesized code is as follows
```

/* Generated by Yosys 0.39+165 (git sha1 22c5ab90d, g++  -fPIC -Os) */

(* src = "rtl.v:1.1-9.10" *)
module top(y, clk);
  (* src = "rtl.v:3.14-3.17" *)
  input clk;
  wire clk;
  (* src = "rtl.v:5.14-5.19" *)
  wire [4:0] wire1;
  (* src = "rtl.v:2.21-2.22" *)
  output [4:0] y;
  wire [4:0] y;
  assign wire1 = 5'h00;
  assign y = 5'h00;
endmodule

```

Its inconsistent output is as follows:
[image]
[image]

Similarly, I have written a script for it to facilitate reproduction.
[yosys_5_14.zip](https://github.com/YosysHQ/yosys/files/15310166/yosys_5_14.zip)

The directory structure of the compressed package is as follows:
rtl.v is the original design. Files with the syn_ prefix represent the synthesized versions. You can compare the vvp.log files located in the identity and yosys directories. Additionally, you can rerun simulations on the synthesized files by executing bash run_sim.sh.

```

yosys_5_14
│
├── data
│   ├── cells_cmos.v
│   ├── cells_cyclone_v.v
│   ├── cells_verific.v
│   ├── cells_xilinx_7.v
│   ├── cells_yosys.v
│
├── identity
│   ├── icarus_stderr.log
│   ├── rtl.v
│   ├── identity_main
│   ├── vvp_identity.log
│   ├── wave_identity.vcd
│   ├── identity_testbench.v
│
├── vivado
│   ├── vivado_testbench.v
│   ├── rtl.v
│   ├── syn_vivado.v
│   ├── icarus_stderr.log
│   ├── vivado_main
│   ├── vvp_vivado.log
│   ├── wave_vivado.vcd
│
├── yosys
│   ├── yosys_testbench.v
│   ├── rtl.v
│   ├── syn_yosys.v
│   ├── icarus_stderr.log
│   ├── yosys_main
│   ├── vvp_yosys.log
│   ├── wave_yosys.vcd
│
├── .Xil
│
├── identity_testbench.v
├── rtl.v
├── run_sim.sh
├── syn_vivado.v
├── syn_yosys.v
├── vivado.jou
├── vivado.log
├── vivado_testbench.v
├── yosys_testbench.v

```

### Expected Behavior

Consistent output before and after synthesis
In order to eliminate interference, I also simulated the Verilog file synthesized by Vivado, and the results were consistent with those before synthesis
[image]

### Actual Behavior

Inconsistent output before and after synthesis

whitequark, 2024-05-14:
Out of curiosity, are you running a fuzzer on Yosys?

KrystalDelusion, 2024-05-14:
This is almost certainly #4395 again. Please stop making new issues for the same problem.

WeneneW, 2024-05-15:
> Out of curiosity, are you running a fuzzer on Yosys?

You're right, I've indeed been doing some fuzzing work recently.Y

WeneneW, 2024-05-15:
> This is almost certainly #4395 again. Please stop making new issues for the same problem.

I apologize for wasting your time. I initially thought these were two different questions. Once again, I'm sorry about that.

closed 2024-05-15 by WeneneW

whitequark, 2024-05-15:
@WeneneW Are you writing a paper?

WeneneW, 2024-05-15:
> @WeneneW Are you writing a paper?

We are working on a fuzz testing tool to enhance the stability of  comprehensive tool. During its operation, the fuzz testing tool will identify potential defects, helping to improve the overall quality of the comprehensive tool.