YosysHQ/yosys #4402: Synthesis issue with taking the negative sign of 0

WeneneW, 2024-05-18:
### Version

Yosys 0.39+165

### On which OS did this happen?

Linux

### Reproduction Steps

Consider the following code. When the input is `wire0 = 6'b111101`, it is a negative number less than `forvar14`. Therefore, `reg16` takes the value of `$unsigned((-forvar9))`, which is `-0`, and the output `y` is 0. However, when using Yosys synthesis, the output result is 1.

```
module top  (y, clk, wire0);
  output wire y;
  input wire clk;

  input wire signed [5:0] wire0;

  reg reg16 = (1'h0);

  reg signed [2:0]forvar14 = (1'h0);

  reg [2:0] forvar10 = (1'h0);
  reg [2:0] forvar9 = (1'h0);

  assign y = reg16;
  always
    @(posedge clk) begin
       for (forvar14 = (1'h0); (forvar14 < (1'h1)); forvar14 = (forvar14 + (1'h1)))
        begin
       reg16 = ($signed((wire0 <= forvar14)) ? $unsigned((-forvar9)) : (^~$signed(forvar10)));
       end
    end
endmodule

```

testbench
```
`include "syn_yosys.v"

`timescale 1ns / 1ps

module testbench;

  reg clk;
  reg signed [5:0] wire0;
  wire y;

  top uut (
    .y(y),
    .clk(clk),
    .wire0(wire0)
  );

  initial begin

    clk = 0;
    wire0 = 6'b111101; 

    forever #5 clk = ~clk;
  end

  initial begin

    #100;
    $display("y = %d", y);
    $finish;
  end

  initial
    begin
      $dumpfile("wave_yosys.vcd");
      $dumpvars(0, testbench);
    end
endmodule

```
Here is the directory structure of the attachment. The `syn_yosys.v` file is the synthesized file. 

In the `identity` directory, execute the following commands:
```
iverilog -o identity_main identity_testbench.v > icarus_stderr.log 2>&1
vvp -n identity_main -lxt2 > vvp_identity.log
```

In the `syn_yosys` directory, execute the following commands:
```
iverilog -o yosys_main yosys_testbench.v > icarus_stderr.log 2>&1
vvp -n yosys_main -lxt2 > vvp_yosys.log
```
to reproduce the issue.
[yosys_5_18.zip](https://github.com/YosysHQ/yosys/files/15365896/yosys_5_18.zip)

```markdown
📂 identity
  ├── 📄 icarus_stderr.log
  ├── 📄 rtl.v
  ├── 📄 identity_testbench.v
  ├── 📄 identity_main
  ├── 📄 vvp_identity.log

📂 syn_yosys
  ├── 📄 rtl.v
  ├── 📄 yosys_testbench.v
  ├── 📄 syn_yosys.v
  ├── 📄 icarus_stderr.log
  ├── 📄 yosys_main
  ├── 📄 vvp_yosys.log
  ├── 📄 wave_yosys.vcd
```

### Expected Behavior

Consistent with the pre synthesis simulation, the result output is 0

### Actual Behavior

Inconsistent output before and after synthesis

KrystalDelusion, 2024-05-19:
Actually I believe the problem is that the output verilog drops the `signed` from wire0, which you can confirm by testing with `wire0 = 6'b0` in the testbench.  The output value is actually being assigned to `^~$signed(forvar10)`, which is where the 1 comes from, not the negative 0.

WeneneW, 2024-05-20:
> Actually I believe the problem is that the output verilog drops the `signed` from wire0, which you can confirm by testing with `wire0 = 6'b0` in the testbench. The output value is actually being assigned to `^~$signed(forvar10)`, which is where the 1 comes from, not the negative 0.
> 
> Compare:
> 
> ```verilog
> assign _01_ = { 1'h0, wire0[4:0] } <= (* src = "docs/test.v:19.25-19.44" *) { wire0[5], 5'h00 };
> ```
> 
> from the verific front end with the read_verilog:
> 
> ```verilog
> assign _06_ = wire0 <= (* src = "docs/test.v:19.26-19.43" *) 3'h0;
> ```

You are correct. I tried the testbench with `wire0 = 6'b0`, and the result was correct, which is the value `0` from `$unsigned((-forvar9))`. However, when I tried the following code with `($signed(wire0)`, the comparison was still incorrect with `wire0 = 6'b111101`. The actual output was the value `1` from `^~$signed(forvar10)`.
﻿
```
module top  (y, clk, wire0);
  output wire y;
  input wire clk;

  input wire signed [5:0] wire0;

  reg reg16 = (1'h0);

  reg signed [2:0]forvar14 = (1'h0);

  reg [2:0] forvar10 = (1'h0);
  reg [2:0] forvar9 = (1'h0);

  assign y = reg16;
  always
    @(posedge clk) begin
       for (forvar14 = (1'h0); (forvar14 < (1'h1)); forvar14 = (forvar14 + (1'h1)))
        begin
       reg16 = ($signed(($signed(wire0) <= forvar14)) ? $unsigned((-forvar9)) : (^~$signed(forvar10)));
       end
    end
endmodule
```

KrystalDelusion, 2026-03-12:
@Noah-S-E wait, you've completely broken your reproducer. Why

gussmith23, 2026-03-24:
@KrystalDelusion were you ever able to reproduce @Noah-S-E 's original issue?

KrystalDelusion, 2026-03-24:
@gussmith23 I had run it locally when I posted my initial reply https://github.com/YosysHQ/yosys/issues/4402#issuecomment-2119460067 and added the bug label, but when I tried more recently I found that the current set of instructions/source are no longer valid (primarily because it refers to `var2` which doesn't exist).  The edit history still has that first version, and at a quick glance I can download the original zip and the code looks right, but I don't remember if I had to do anything else to get it to work locally.  Lemme know if it would be helpful for me to do a pass over it and minimize properly.