YosysHQ/yosys #4403: make error 'abc' is not configured as a git submodule.

Krishnakumarmohanraj, 2024-05-20:
### Version

latest

### On which OS did this happen?

Linux

### Reproduction Steps

[Makefile.conf] CONFIG := gcc
Error: 'abc' is not configured as a git submodule.
To resolve this:
1. Back up your changes: Save any modifications from the 'abc' directory to another location.
2. Remove the existing 'abc' directory: Delete the 'abc' directory and all its contents.
3. Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule.
4. Reapply your changes: Move your saved changes back to the 'abc' directory, if necessary.
make: *** [Makefile:748: check-git-abc] Error 1

### Expected Behavior

When i give make, its shows to remove abc folder and asked to do git submodule update --init, we done several times but same error again

### Actual Behavior

Can you give solution for it please

RCoeurjoly, 2024-05-20:
When you execute `git submodule update --init`, is there any error message?

Krishnakumarmohanraj, 2024-05-20:
Hi, git submodule dont show any error, but once clone the submodule, and do --> make, its showing error as below

[root@coeserver yosys]# make
[Makefile.conf] CONFIG := gcc
Error: 'abc' is not configured as a git submodule.
To resolve this:
1. Back up your changes: Save any modifications from the 'abc' directory to another location.
2. Remove the existing 'abc' directory: Delete the 'abc' directory and all its contents.
3. Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule.
4. Reapply your changes: Move your saved changes back to the 'abc' directory, if necessary.
make: *** [Makefile:748: check-git-abc] Error 1
[root@coeserver yosys]# **rm -rf abc**
[root@coeserver yosys]# **git submodule update --init**
Submodule path 'abc': checked out '237d81397fcc85dd3894bf1a449d2955cd3df02d'
[root@coeserver yosys]# **make**
[Makefile.conf] CONFIG := gcc
Error: 'abc' is not configured as a git submodule.
To resolve this:
1. Back up your changes: Save any modifications from the 'abc' directory to another location.
2. Remove the existing 'abc' directory: Delete the 'abc' directory and all its contents.
3. Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule.
4. Reapply your changes: Move your saved changes back to the 'abc' directory, if necessary.
make: *** [Makefile:748: check-git-abc] Error 1
[root@coeserver yosys]#

Krishnakumarmohanraj, 2024-05-20:
[root@coeserver yosys]# git --version
git version 1.8.3.1
[root@coeserver yosys]# cat /etc/redhat-release 
CentOS Linux release 7.9.2009 (Core)

RCoeurjoly, 2024-05-20:
git -C <path> was added in git 1.8.5, https://github.com/git/git/blob/master/Documentation/RelNotes/1.8.5.txt#L115.
Can you update git to at least 1.8.5?

salfter, 2024-06-07:
I'm running into the same problem, and I'm using git 2.44.2.

rowanG077, 2024-06-07:
Are you sure you have the submodules?

salfter, 2024-06-08:
> Are you sure you have the submodules?

Yes.  I'm trying to update my Gentoo ebuild.  I've already switched it over from building from tarballs to building from Git, and the default behavior of git-r3.eclass is to recursively check out submodules.  

By "zeroing out" the ```check-git-abc``` function in the makefile (replace the contents with ```exit 0;```), I was able to get Yosys to build.

davekeeshan, 2024-06-08:
I see this when I download the [tar.gz file from the tag](https://github.com/YosysHQ/yosys/archive/refs/tags/yosys-0.42.tar.gz) (which is how [spack ](https://github.com/spack/spack)installs releases from), from github, which isn't a git repo, when the makefile finds an empty abc/ directory, it issues the above error, but one cannot do a submodule update on a directory that is not a git repo.

RCoeurjoly, 2024-06-10:
> I see this when I download the [tar.gz file from the tag](https://github.com/YosysHQ/yosys/archive/refs/tags/yosys-0.42.tar.gz) (which is how [spack ](https://github.com/spack/spack)installs [...]

In that case, I think the best way to proceed would be to download the corresponding abc from [here](https://github.com/YosysHQ/abc/tags). Then decompress into abc directory.
The Makefile should be able to work on those cases, yosys and abc as git repos, and yosys and abc as tar.gz

davekeeshan, 2024-06-10:
> In that case, I think the best way to proceed would be to download the corresponding abc from [here](https://github.com/YosysHQ/abc/tags). Then decompress into abc directory. The Makefile should be [...]

I know what you are saying, but for me that seems to defy the purpose of a Makefile flow, if the user has to "do something" extra, you start to lose people. I personally know how to go around this, but this is a observation on general flow.

A simple fix is to set the default case to not compile `abc` at all, and `abc` is something extra.

As a reference, I am maintaining the compile of `yosys` in the tool [`spack`](https://github.com/spack/spack) (also icarus, verible and verilator) and it is in this context that the compile is falling over

[yosys/package.py](
https://github.com/spack/spack/blob/8f0c28037b92188e3b88a0166366a69d80fd11be/var/spack/repos/builtin/packages/yosys/package.py#L9)

RCoeurjoly, 2024-06-13:
> > > I see this when I download the [tar.gz file from the tag](https://github.com/YosysHQ/yosys/archive/refs/tags/yosys-0.42.tar.gz) (which is how [spack ](https://github.com/spack/spack)installs releases from), from github, which isn't a git repo, when the makefile finds an empty `abc/` directory, it issues the above error, but one cannot do a submodule update on a directory that is not a git repo.
> > 
> > 
> > In that case, I think the best way to proceed would be to download the corresponding abc from [here](https://github.com/YosysHQ/abc/tags). Then decompress into abc directory. The Makefile should be able to work on those cases, yosys and abc as git repos, and yosys and abc as tar.gz
> 
> I know what you are saying, but for me that seems to defy the purpose of a Makefile flow, if the user has to "do something" extra, you start to lose people. I personally know how to go around this, but this is a observation on general flow.
> 
> A simple fix is to set the default case to not compile `abc` at all, and `abc` is something extra. Alternatively you could put a wget/curl of the extra repo into the flow.
> 
> As a reference, I am maintaining the compile of `yosys` in the tool [`spack`](https://github.com/spack/spack) (also icarus, verible and verilator) and it is in this context that the compile is falling over
> 
> [yosys/package.py](https://github.com/spack/spack/blob/8f0c28037b92188e3b88a0166366a69d80fd11be/var/spack/repos/builtin/packages/yosys/package.py#L9)

I understand your frustration. For reference, before we were executing git clone inside the Makefile. This was change to a git submodule and better error messages, which you are seeing now.

I see in yosys/package.py you set ENABLE_ABC if there is an abc dir (if I understand the script correctly). That was exactly the solution I wanted to propose

davekeeshan, 2024-06-13:
> I understand your frustration. For reference, before we were executing git clone inside the Makefile. This was change to a git submodule and better error messages, which you are seeing now.
>
> I see [...]

Thanks for the feedback.

I have been able to solve it inside the environment I was using, so I am not personally blocked by this.

I still haven't figured out how important `abc` is for a general build, does having it disabled as a default make more sense?

However, my comments here were more for the general case, where someone is working with the release tar file.  I can see how a `git checkout` would work, as it is discrete and independent.  However a `git submodule` can only be executed from inside a correctly setup git repo so don't be surprised if random people report this as a issue from now on (or hopefully they just find the content of this issue)

nakengelhardt, 2024-06-14:
> I still haven't figured out how important `abc` is for a general build, does having it disabled as a default make more sense?

ABC is almost mandatory for any synthesis flow, it's the only pass that can reliably map general logic to the target architecture primitives (LUTs or gates). You can still use CXXRTL and custom flows, but without ABC Yosys will be useless for the majority of users.

whitequark, 2024-06-14:
To add to this, there is the `-noabc` flow for iCE40 FPGAs, but it is largely inferior to the ABC flow in terms of area and delay. While perhaps not _useless_, its only real advantage is a slightly improved debugging experience (it preserves more intermediate net names), which is not worth the degraded delay/area for most users.

ElecGeek, 2024-10-06:
I had the problem in a GitHub workflow as well:
The git submodule update init was before, rather than after the cd <yosys folder> (more the mistake is big, less we can see it).

As a workaround, this can be fixed WITHOUT disabling abc. When one is in the Yosys source folder, ensure the abc folder is empty and git clone berkeley-abc/abc to the folder abc (or untar a tarball). Of course, one has to do it again for updates.