YosysHQ/yosys #4410: create a pip install package wheel for Windows

eabase, 2024-05-22:
### Feature Description

We're trying to gain some traction in the open GPU project, [VeriGPU](https://github.com/hughperkins/VeriGPU), which depends on this project. 

It would be great if you could create a PyPi pip package building the wheels for native windows and WSL.

whitequark, 2024-05-23:
What specifically do you want the wheels to do?

eabase, 2024-05-24:
@whitequark 
> What specifically do you want to do with the wheels?

That's how you create installable packages in python.

whitequark, 2024-05-24:
I know what a wheel is. I'm asking what you want to do with Yosys packaged as a wheel.

donn, 2024-07-22:
Not OP but also interested in this-- it would let Python packages distributed over PIP depend on pyosys (and then both can be grabbed with a `pip install`)

eabase, 2024-08-09:
> I know what a wheel is. I'm asking what you want to do with Yosys packaged as a wheel.

I'm not sure you do, since you are asking.  It allows you to distribute pre-compiled binaries that can be used out-of-the-box on any OS/system. 

There is no good reason that every single Windows, *nix or Mac owner should need to compile their own binaries, unless you find some kind of pleasure watching compilation logs scroll by your screen for several minutes.

eabase, 2024-08-09:
Eventually, we can do the same for items here:
https://github.com/hughperkins/VeriGPU/blob/main/docs/installation.md