YosysHQ/yosys #4411: Enhancing Syntax Checking for Port Declarations in Yosys

WeneneW, 2024-05-23:
### Version

Yosys 0.39+165

### On which OS did this happen?

Linux

### Reproduction Steps

In Yosys, the `read_verilog` and `synth` commands can be used to read and synthesize Verilog code. However, they may not always detect certain syntax errors in the port declaration list. For example, consider the following code:

```verilog
module top (
  output wire y,
  input wire clk,
);

  wire [2:0] wire1 = 3'b000;
  assign y = wire1; 
endmodule
```

According to the Verilog language specification, the `comma`after `input wire clk,` should not be there. Normally, this should be a syntax error. However, when I use Yosys to parse and synthesize this code, it does not detect this error and synthesizes successfully.

Is it possible to enhance the syntax checking for port declarations in Yosys using the `read_verilog` and `synth` commands?

### Expected Behavior

`read_verilog` can report syntax errors.

### Actual Behavior

There was no error report.

zachjs, 2024-05-26:
Per the README, the Yosys Verilog frontend intentionally allows trailing commas. I think many Yosys users already depend on this trailing comma behavior (enough that I've had to tolerate trailing commas in sv2v, too). As such, I don't think we can change the default behavior of `read_verilog` to forbid trailing commas. Perhaps we could have a stricter parsing mode, but I'm not sure it would get much use. Some users also choose to pass their source through other tools such as verilator or iverilog for perform additional checks.

whitequark, 2024-05-27:
This doesn't seem like a valuable enhancement; to me it looks like a padding for your fuzzer's list of discoveries.

closed 2024-05-27 by whitequark