YosysHQ/yosys #4419: CXXRTL: >20x compile time regression with clang++-18

Wren6991, 2024-05-27:
### Version

Yosys 0.39+4 (git sha1 3231c1cd9, clang++ 16.0.6 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

This zip file contains the full dut.cpp from `write_cxxrtl`, which shows a 24x compile time regression:
[dut-full.zip](https://github.com/YosysHQ/yosys/files/15454860/dut-full.zip)

Or, this zip file contains a reduced dut.cpp (900k instead of 9M), which shows a 6x compile time regression:
[dut-reduced.zip](https://github.com/YosysHQ/yosys/files/15454862/dut-reduced.zip)

Good:

```bash
time clang++-17 -O3 -c -std=c++14 -I $(yosys-config --datdir)/include/backends/cxxrtl/runtime dut.cpp
```

Bad:

```bash
time clang++-18 -O3 -c -std=c++14 -I $(yosys-config --datdir)/include/backends/cxxrtl/runtime dut.cpp
```

The full dut.cpp can be reproduced from Verilog source by:

```bash
git clone git@github.com:Wren6991/Hazard3.git hazard3
cd hazard3
git checkout 5b31e2679
git submodule update --init -- scripts
. sourceme
cd test/sim/tb_cxxrtl
make
```

The reduced DUT has the hierarchy trimmed to just the `hazard3_decode` module, and all optional instruction extensions disabled.

This reproduces at -Og and above, but not at -O0.

### Expected Behavior

Compile takes 45 seconds, as with clang++-16, clang++-17.

### Actual Behavior

Compile takes 18 minutes with clang++-18.

I dithered on whether to report this here, but it seems like there is something about CXXRTL's code generation that hits some pathologically slow case in clang++-18, and maybe this could be improved from the CXXRTL side.

Alternatively if there is some recommended set of clang/llvm flags to use with CXXRTL to disable problematic passes and stop the compile time from blowing up, documenting that would also be helpful.

clang++-18 is the default clang++ as of Ubuntu 24.04 LTS, so I imagine more people will start hitting this.

whitequark, 2024-05-27:
Oh wow, that hurts.

I think there is almost certainly an LLVM/Clang regression underlying. Can you profile it please?

Wren6991, 2024-06-01:
Here is the output from `-ftime-report`. First for clang++-17 (good):

<details>

```
clang++-17 -ftime-report -O3 -std=c++14  -I /usr/local/share/yosys/include/backends/cxxrtl/runtime -I build-tb tb.cpp -o tb
===-------------------------------------------------------------------------===
                          Pass execution timing report
===-------------------------------------------------------------------------===
  Total Execution Time: 30.6422 seconds (30.6881 wall clock)

   ---User Time---   --System Time--   --User+System--   ---Wall Time---  --- Name ---
  16.1551 ( 53.7%)   0.0449 (  8.0%)  16.2000 ( 52.9%)  16.2014 ( 52.8%)  SROAPass
   5.1537 ( 17.1%)   0.1296 ( 23.0%)   5.2834 ( 17.2%)   5.2919 ( 17.2%)  InstCombinePass
   2.0826 (  6.9%)   0.0125 (  2.2%)   2.0951 (  6.8%)   2.0950 (  6.8%)  JumpThreadingPass
   1.9233 (  6.4%)   0.1124 ( 20.0%)   2.0357 (  6.6%)   2.0483 (  6.7%)  InlinerPass
   1.3596 (  4.5%)   0.0224 (  4.0%)   1.3821 (  4.5%)   1.3815 (  4.5%)  CorrelatedValuePropagationPass
   0.5307 (  1.8%)   0.0081 (  1.4%)   0.5388 (  1.8%)   0.5402 (  1.8%)  IPSCCPPass
   0.5058 (  1.7%)   0.0223 (  4.0%)   0.5282 (  1.7%)   0.5277 (  1.7%)  GVNPass
   0.4481 (  1.5%)   0.0391 (  6.9%)   0.4872 (  1.6%)   0.4944 (  1.6%)  EarlyCSEPass
   0.3813 (  1.3%)   0.0320 (  5.7%)   0.4133 (  1.3%)   0.4133 (  1.3%)  SimplifyCFGPass
   0.2193 (  0.7%)   0.0000 (  0.0%)   0.2193 (  0.7%)   0.2202 (  0.7%)  GlobalOptPass
   0.2147 (  0.7%)   0.0000 (  0.0%)   0.2147 (  0.7%)   0.2148 (  0.7%)  CalledValuePropagationPass
   0.1690 (  0.6%)   0.0006 (  0.1%)   0.1696 (  0.6%)   0.1773 (  0.6%)  SLPVectorizerPass
   0.1158 (  0.4%)   0.0000 (  0.0%)   0.1158 (  0.4%)   0.1168 (  0.4%)  RequireAnalysisPass<llvm::GlobalsAA, llvm::Module, llvm::AnalysisManager<Module>>
   0.0669 (  0.2%)   0.0219 (  3.9%)   0.0888 (  0.3%)   0.0890 (  0.3%)  PostOrderFunctionAttrsPass
   0.0670 (  0.2%)   0.0072 (  1.3%)   0.0741 (  0.2%)   0.0739 (  0.2%)  DSEPass
   0.0666 (  0.2%)   0.0021 (  0.4%)   0.0688 (  0.2%)   0.0689 (  0.2%)  CallSiteSplittingPass
   0.0524 (  0.2%)   0.0091 (  1.6%)   0.0615 (  0.2%)   0.0611 (  0.2%)  ReassociatePass
   0.0502 (  0.2%)   0.0070 (  1.2%)   0.0573 (  0.2%)   0.0571 (  0.2%)  SCCPPass
   0.0457 (  0.2%)   0.0098 (  1.7%)   0.0555 (  0.2%)   0.0553 (  0.2%)  ADCEPass
   0.0416 (  0.1%)   0.0071 (  1.3%)   0.0487 (  0.2%)   0.0486 (  0.2%)  BDCEPass
   0.0355 (  0.1%)   0.0104 (  1.9%)   0.0459 (  0.1%)   0.0460 (  0.2%)  LoopSimplifyPass
   0.0340 (  0.1%)   0.0037 (  0.7%)   0.0376 (  0.1%)   0.0378 (  0.1%)  LowerExpectIntrinsicPass
   0.0289 (  0.1%)   0.0044 (  0.8%)   0.0333 (  0.1%)   0.0332 (  0.1%)  MemCpyOptPass
   0.0279 (  0.1%)   0.0000 (  0.0%)   0.0279 (  0.1%)   0.0279 (  0.1%)  AssignmentTrackingPass
   0.0221 (  0.1%)   0.0036 (  0.6%)   0.0257 (  0.1%)   0.0258 (  0.1%)  PromotePass
   0.0190 (  0.1%)   0.0035 (  0.6%)   0.0226 (  0.1%)   0.0224 (  0.1%)  ConstraintEliminationPass
   0.0154 (  0.1%)   0.0045 (  0.8%)   0.0198 (  0.1%)   0.0199 (  0.1%)  LCSSAPass
   0.0163 (  0.1%)   0.0031 (  0.5%)   0.0194 (  0.1%)   0.0192 (  0.1%)  TailCallElimPass
   0.0155 (  0.1%)   0.0022 (  0.4%)   0.0176 (  0.1%)   0.0176 (  0.1%)  IndVarSimplifyPass
   0.0147 (  0.0%)   0.0023 (  0.4%)   0.0170 (  0.1%)   0.0169 (  0.1%)  LICMPass
   0.0129 (  0.0%)   0.0038 (  0.7%)   0.0167 (  0.1%)   0.0166 (  0.1%)  LoopRotatePass
   0.0156 (  0.1%)   0.0000 (  0.0%)   0.0156 (  0.1%)   0.0156 (  0.1%)  RecomputeGlobalsAAPass
   0.0121 (  0.0%)   0.0029 (  0.5%)   0.0150 (  0.0%)   0.0149 (  0.0%)  AggressiveInstCombinePass
   0.0105 (  0.0%)   0.0038 (  0.7%)   0.0142 (  0.0%)   0.0143 (  0.0%)  RequireAnalysisPass<llvm::ShouldNotRunFunctionPassesAnalysis, llvm::Function, llvm::AnalysisManager<Function>>
   0.0110 (  0.0%)   0.0022 (  0.4%)   0.0132 (  0.0%)   0.0132 (  0.0%)  LoopIdiomRecognizePass
   0.0131 (  0.0%)   0.0000 (  0.0%)   0.0131 (  0.0%)   0.0131 (  0.0%)  InstSimplifyPass
   0.0088 (  0.0%)   0.0021 (  0.4%)   0.0110 (  0.0%)   0.0109 (  0.0%)  VectorCombinePass
   0.0073 (  0.0%)   0.0033 (  0.6%)   0.0105 (  0.0%)   0.0105 (  0.0%)  InvalidateAnalysisPass<llvm::AAManager>
   0.0084 (  0.0%)   0.0000 (  0.0%)   0.0084 (  0.0%)   0.0084 (  0.0%)  ReversePostOrderFunctionAttrsPass
   0.0074 (  0.0%)   0.0011 (  0.2%)   0.0084 (  0.0%)   0.0084 (  0.0%)  LoopFullUnrollPass
   0.0067 (  0.0%)   0.0013 (  0.2%)   0.0080 (  0.0%)   0.0080 (  0.0%)  Float2IntPass
   0.0078 (  0.0%)   0.0000 (  0.0%)   0.0078 (  0.0%)   0.0078 (  0.0%)  LoopUnrollPass
   0.0070 (  0.0%)   0.0006 (  0.1%)   0.0077 (  0.0%)   0.0076 (  0.0%)  LoopDeletionPass
   0.0073 (  0.0%)   0.0000 (  0.0%)   0.0073 (  0.0%)   0.0073 (  0.0%)  GlobalDCEPass
   0.0000 (  0.0%)   0.0003 (  0.1%)   0.0003 (  0.0%)   0.0069 (  0.0%)  CGProfilePass
   0.0051 (  0.0%)   0.0016 (  0.3%)   0.0067 (  0.0%)   0.0066 (  0.0%)  LibCallsShrinkWrapPass
   0.0046 (  0.0%)   0.0017 (  0.3%)   0.0063 (  0.0%)   0.0063 (  0.0%)  CoroSplitPass
   0.0045 (  0.0%)   0.0016 (  0.3%)   0.0061 (  0.0%)   0.0062 (  0.0%)  CoroElidePass
   0.0046 (  0.0%)   0.0016 (  0.3%)   0.0062 (  0.0%)   0.0061 (  0.0%)  OpenMPOptCGSCCPass
   0.0044 (  0.0%)   0.0016 (  0.3%)   0.0061 (  0.0%)   0.0060 (  0.0%)  SpeculativeExecutionPass
   0.0044 (  0.0%)   0.0015 (  0.3%)   0.0059 (  0.0%)   0.0059 (  0.0%)  MoveAutoInitPass
[... cut ...]
  Total Execution Time: 1054.1768 seconds (1054.4013 wall clock)

   ---User Time---   --System Time--   --User+System--   ---Wall Time---  --- Name ---
  1051.2682 (100.0%)   2.9086 (100.0%)  1054.1768 (100.0%)  1054.4013 (100.0%)  Clang front-end timer
  1051.2682 (100.0%)   2.9086 (100.0%)  1054.1768 (100.0%)  1054.4013 (100.0%)  Total
```

</details>

So almost all of the time is spent in LoopRotatePass (2/3rds) and SROAPass (1/3rd). LoopRotatePass seems like it may not be that useful for the straight line code generated by CXXRTL?

whitequark, 2024-06-01:
Yep! This seems like a straightforward bug. I should be able to fix it whenever I get the chance, but I'm not sure when exactly that will happen.