YosysHQ/yosys #4435: cell reference names appear to be truncated

Cronus-38, 2024-06-07:
### Version

Yosys 0.35+7 (git sha1 5691cd095, clang 15.0.7 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

I can't say with certainty that the behavior I'm seeing is a bug, but I can't find any explanation for the tool's behavior.  I'm running with a proprietary library, so I won't be able to supply a testcase that demonstrates the issue.  Instead, I've provided excerpts from the .lib file as well as the synthesized netlist in the hope that you can explain what's happening.

### Expected Behavior

The library .lib contains cells such as this:

  cell (AOI21_X1F_A7P5PP84TL_C14) {...

I can see that the .lib file is being read in correctly, and it is the ONLY .lib being read into the tool.

### Actual Behavior

The synthesized netlist contains cells such as this:

  AOI21_X1 _13708_ (...

Any insight into this is appreciated!  Oh, and I should mention that I'm runnin Yosys via the OpenRoad flow.  Thanks!

widlarizer, 2024-07-11:
AOI21_X1 is a cell present in the nangate cell library. You should check the yosys invocation that OpenROAD uses. It's likely you're not actually using the liberty file for your cell library. Search for the `read_liberty` yosys pass in your logs. Other than this hint, not much we can do for you

Cronus-38, 2024-07-11:
Thanks for the reply.  Earlier this week I reached the same conclusion:  the cells were coming from the nangate45 library.  I had previously run the flow in the same directory using the nangate45 library.  It's still not clear to me why Yosys was loading the nangate45 library since it didn't appear anywhere in the log, so perhaps the messaging could be improved in that regard.  At any rate, I got it working by running in a clean directory, so please consider this issue closed.  Thanks again.

closed 2024-07-11 by Cronus-38