YosysHQ/yosys #4442: Stub RTL Generation Feature

Peter-Herrmann, 2024-06-10:
### Feature Description

Designs frequently need to stub out modules to simplify simulation or synthesis during development. As part of workflows, it is very useful to be able to automatically generate stub modules from design files.
 
I would like to request a feature where Yosys can "stub-out" a module, so that `write-verilog` would produce a stub module. 

**Bare Minimum Features**

- Remove all internal logic
- Tie all outputs to zero

**Nice-to-have Features**

- Configurable stub output behavior:
  - Selectively assign constants to outputs
  - Pass-through ports
  - Flop (N-cycle delay)

widlarizer, 2024-07-11:
Can you further describe your use case? Is this not feasible to do and control in your sources?

povik, 2024-07-15:
@Peter-Herrmann From a preliminary discussion we are unlikely to support a full-fledged `stub` command which would provide all of the nice-to-have features.

povik, 2024-07-15:
@Peter-Herrmann From a preliminary discussion we are unlikely to support a full-fledged stub command which would provide all of the nice-to-have features. Some elementary stubbing can be achieved with existing Yosys commands, and those we can extend if there's an important omission, but once the stubbing involves the generation of non-trivial logic, we prefer users write HDL for those cases over a specialized `stub` command.

That being said I think you should be able to stub-out `foo` and connect the `B_o` output to a constant with the following:

    delete top/* top/x:* %d
    connect -set B_o 0 foo

The first command deletes everything but the ports. The second connects a constant. I am not sure if there's a command to create a pass-through.

Peter-Herrmann, 2024-07-15:
The same effect could be achieved in the source code with the preprocessor, but that has a few problems at scale. 

- For very large designs, this can be extremely tedious, and custom automation scripts are the usual solution. Instead of everyone having custom scripts to automate stub generation, it would be nice to have at least one popular tool that has a natively supported solution.
- It is often necessary to stub out third party IPs that we do not want to modify directly (wrappers can achieve this, but it clutters the design hierarchy and can introduce breaking changes to existing testbenches)
- It can dramatically decrease readability of source files, especially for large designs with many ports. Having every module come with stubbing-out lines could as much as double the size of the code base.
- Having a tool to generate stubs procedurally reduces maintenance (port changes don't need to be changed in every configuration) and reduces surface area for bugs.

Peter-Herrmann, 2024-07-15:
> once the stubbing involves the generation of non-trivial logic, we prefer users write HDL for those cases over a specialized `stub` command.

I think this is a very reasonable stance.

> Some elementary stubbing can be achieved with existing Yosys commands

I will try this out, this may be enough for my needs. Thank you!

closed 2024-07-15 by Peter-Herrmann

povik, 2024-07-15:
I noticed I have mistyped the sample commands (the module name would be `foo` not `top`), see now