YosysHQ/yosys #4447: system verilog "import" command not recognized (ERROR: syntax error, unexpected TOK_ID)

titan73, 2024-06-11:
### Version

Yosys 0.42+10 (git sha1 ef9045882, clang++ 14.0.0-1ubuntu1.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Compile a system verilog file with only "import foobar;" or "import foobar::*;"

### Expected Behavior

Compile with no error

### Actual Behavior

yosys> read_verilog -sv toto.sv

1. Executing Verilog-2005 frontend: foobar.sv
Parsing SystemVerilog input from `foobar.sv' to AST representation.
foobar.sv:1: ERROR: syntax error, unexpected TOK_ID

titan73, 2024-06-11:
Oddly a file containing a package (package keyword) is compiled perfectly.

whitequark, 2024-06-11:
Yosys does not support SystemVerilog imports at the moment, and as far as I know there are no plans to. You can use sv2v, or purchase a commercial license which lets you use the Verific frontend that has SV support.

titan73, 2024-06-18:
Ok I used sv2v and it worked. Thanks!

closed 2024-06-18 by titan73