YosysHQ/yosys #4449: check-git-abc in Makefile for the first compile time

nlwmode, 2024-06-13:
### Version

0.41

### On which OS did this happen?

Linux

### Reproduction Steps

1. git clone --recursive https://github.com/YosysHQ/yosys.git
2. cd yosys
3. mkdir build && cd build
4. make -j 128

### Expected Behavior

  Build successful.

### Actual Behavior

Error: 'abc' is not configured as a git submodule.";
To resolve this:";
1. Back up your changes: Save any modifications from the 'abc' directory to another location.";
2. Remove the existing 'abc' directory: Delete the 'abc' directory and all its contents.";
3. Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule.";
4. Reapply your changes: Move your saved changes back to the 'abc' directory, if necessary.";

nlwmode, 2024-06-13:
> @nlwmode
> 
> Last step in this case should be `make -C .. -j 128` since makefile is in directory above, but that still does not explain issue you get. Could you please give us info about output of `git --version` on your machine ?

"git version 2.42.0".

The PR (https://github.com/YosysHQ/yosys/pull/4450) carefully explains and solves the problem. Please checking that.

RCoeurjoly, 2024-06-18:
I cannot reproduce

This is what I have done:

```
git clone --recursive https://github.com/YosysHQ/yosys.git yosys_4449
cd yosys_4449/
mkdir build && cd build
make -C ..
```
I have git version 2.43.1
Can you execute these commands to see if you still have the issue?

kittennbfive, 2024-06-27:
Hi,

i can confirm that something is wrong and the output of `make install` (do this, do this) does not work.

I tried this:
```
$ cd /tmp
$ git clone --recursive https://github.com/YosysHQ/yosys.git yosys_4449
$ cd yosys_4449/
$ mkdir build && cd build
$ make -C .. -j12
$ su -
# cd /tmp/yosys_4449
# make install
```

and got:
```
Error: 'abc' is not configured as a git submodule.
To resolve this:
1. Back up your changes: Save any modifications from the 'abc' directory to another location.
2. Remove the existing 'abc' directory: Delete the 'abc' directory and all its contents.
3. Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule.
4. Reapply your changes: Move your saved changes back to the 'abc' directory, if necessary.
```
(But as i said doing so does not solve the issue.)

I am using git version 2.39.2 on Debian stable (12.5 currently).

**Workaround**
At Makefile:770 remove the entire "if" and replace by exit 0; . Installation works and Yosys works fine too.

Thank you!

janrinze, 2024-09-17:
#4450 appears to work for me on Debian 12.
Could this be a Debian specific issue?