YosysHQ/yosys #4451: `count_id(cell->name) == 0` failure

wyager, 2024-06-13:
### Version

Yosys 0.42+12 (git sha1 a55e8594b, aarch64-apple-darwin21.4-clang++ 14.0.0-1ubuntu1.1 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

Download this zip file and run `run_me.sh` (not a virus, I promise)
[error-repro.zip](https://github.com/user-attachments/files/15828849/error-repro.zip)

I tried running with `-dff` per @Ravenslofty's suggestion, no dice

### Expected Behavior

yosys should synthesize something

### Actual Behavior

```
ERROR: Assert `count_id(cell->name) == 0' failed in kernel/rtlil.cc:2152.
```

Ravenslofty, 2024-06-15:
This is the same root cause as #4349 but instead of wire naming, it's cell naming.

In the Dev JF it was proposed that synthesis scripts call `splitnets` for FPGAs, much like the external ASIC synthesis scripts do. That fixes #4349, and it also fixes this issue.

Hopefully I remember to submit a PR to that effect.

wyager, 2024-11-28:
Anything I can do to help on this issue? Would like to pick this up again and currently blocked on this one step

tmolteno, 2025-08-08:
Same error here when synthesizing the icesugar Pro blink example. https://github.com/wuxx/icesugar-pro/tree/master/src/blink. 

```
yowasp-yosys --version
Yosys 0.56 (git sha1 9c447ad9d, ccache clang++ 18.1.3 -O3 -flto -flto)
```

```
Extracted 31 AND gates and 196 wires from module `blink' to a netlist network with 28 inputs and 32 outputs.

5.39.15.4. Executing ABC9_EXE pass (technology mapping using ABC9).

5.39.15.5. Executing ABC9.
Running ABC command: "/yosys-abc" -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_lut <abc-temp-dir>/input.lut 
ABC: + read_box <abc-temp-dir>/input.box 
ABC: + &read <abc-temp-dir>/input.xaig 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =     28/     32  and =      29  lev =    6 (0.14)  mem = 0.00 MB  box = 11  bb = 0
ABC: + &scorr 
ABC: Warning: The network is combinational.
ABC: + &sweep 
ABC: + &dc2 
ABC: + &dch -f 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =     28/     32  and =      53  lev =    6 (0.14)  mem = 0.00 MB  ch =    7  box = 11  bb = 0
ABC: cst =       0  cls =      7  lit =       7  unused =     100  proof =     0
ABC: + &if -W 300 -v 
ABC: K = 7. Memory (bytes): Truth =    0. Cut =   64. Obj =  120. Set =  624. CutMin = no
ABC: Node =      53.  Ch =     7.  Total mem =    0.03 MB. Peak cut mem =    0.01 MB.
ABC: P:  Del = 1634.00.  Ar =      29.0.  Edge =       45.  Cut =      411.  T =     0.00 sec
ABC: P:  Del = 1634.00.  Ar =      29.0.  Edge =       45.  Cut =      411.  T =     0.00 sec
ABC: P:  Del = 1634.00.  Ar =      29.0.  Edge =       45.  Cut =      521.  T =     0.00 sec
ABC: F:  Del = 1597.00.  Ar =      17.0.  Edge =       41.  Cut =      445.  T =     0.00 sec
ABC: A:  Del = 1597.00.  Ar =      15.0.  Edge =       42.  Cut =      220.  T =     0.00 sec
ABC: A:  Del = 1597.00.  Ar =      15.0.  Edge =       42.  Cut =      205.  T =     0.00 sec
ABC: Total time =     0.00 sec
ABC: + &write -n <abc-temp-dir>/output.aig 
ABC: + &mfs 
ABC: The network is not changed by "&mfs".
ABC: + &ps -l 
ABC: <abc-temp-dir>/input : i/o =     28/     32  and =      35  lev =    6 (0.14)  mem = 0.00 MB  box = 11  bb = 0
ABC: Mapping (K=6)  :  lut =     11  edge =      42  lev =    3 (0.07)  levB =   11  mem = 0.00 MB
ABC: LUT = 11 : 2=1 9.1 %  3=3 27.3 %  4=5 45.5 %  5=1 9.1 %  6=1 9.1 %  Ave = 3.82
ABC: + &write -n <abc-temp-dir>/output.aig 
ABC: + time 
ABC: elapse: 0.01 seconds, total: 0.01 seconds

5.39.15.6. Executing AIGER frontend.
M=96 I=61 L=0 O=131 A=35
ERROR: Assert `count_id(wire->name) == 0' failed in ../yosys-src/kernel/rtlil.cc:2621.
make: *** [Makefile:9: blink.json] Error 1
```

KrystalDelusion, 2025-08-15:
@tmolteno I can't reproduce your error with a regular build of Yosys, so I suspect yours is a case of #5274 rather than this issue.

I've also minimized this issue to the following RTLIL:
```
# Generated by Yosys 0.56+105 (git sha1 ec18d1aed, ccache clang++ 18.1.3 -fPIC -O3)
autoidx 2948
attribute \top 1
module \orange_crab
  wire input 9 $auto$bugpoint.cc:309:simplify_something$1244
  wire width 36 input 10 $auto$bugpoint.cc:309:simplify_something$1313
  wire input 11 $auto$bugpoint.cc:309:simplify_something$2108
  wire input 12 $auto$bugpoint.cc:309:simplify_something$2436
  wire width 4 output 13 $auto$bugpoint.cc:309:simplify_something$2446
  wire input 14 $auto$bugpoint.cc:309:simplify_something$2508
  wire width 4 input 15 $auto$bugpoint.cc:309:simplify_something$2625
  wire width 28 output 1 $auto$bugpoint.cc:309:simplify_something$841
  wire width 24 input 2 $auto$bugpoint.cc:309:simplify_something$858
  wire width 29 input 3 $auto$bugpoint.cc:309:simplify_something$862
  wire width 29 output 4 $auto$bugpoint.cc:309:simplify_something$865
  wire width 29 output 5 $auto$bugpoint.cc:309:simplify_something$873
  wire width 28 output 6 $auto$bugpoint.cc:309:simplify_something$877
  wire width 29 output 7 $auto$bugpoint.cc:309:simplify_something$921
  wire width 37 input 8 $auto$bugpoint.cc:309:simplify_something$941
  wire $auto$opt_dff.cc:247:make_patterns_logic$2410
  wire $auto$opt_dff.cc:272:make_patterns_logic$2412
  wire width 9 $delete_wire$2833
  wire $delete_wire$2834
  wire $procmux$1580_CMP
  wire $procmux$1581_CMP
  wire width 39 \c$case_alt_50
  wire width 38 \c$ds_app_arg_21
  wire width 38 \c$ds_case_alt_24
  wire width 42 \ds_3
  wire width 43 \result_62
  cell $dffe $auto$ff.cc:266:slice$2409
    parameter \CLK_POLARITY 1
    parameter \EN_POLARITY 1
    parameter \WIDTH 4
    connect \CLK $auto$bugpoint.cc:309:simplify_something$2508
    connect \D $auto$bugpoint.cc:309:simplify_something$2625
    connect \EN $auto$opt_dff.cc:272:make_patterns_logic$2412
    connect \Q $auto$bugpoint.cc:309:simplify_something$2446
  end
  cell $adff $auto$ff.cc:266:slice$2806
    parameter \ARST_POLARITY 1
    parameter \ARST_VALUE 35'0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    parameter \CLK_POLARITY 1
    parameter \WIDTH 35
    connect \ARST $auto$bugpoint.cc:309:simplify_something$2108
    connect \CLK $auto$bugpoint.cc:309:simplify_something$2436
    connect \D { \result_62 [37:35] \result_62 [31:0] }
    connect \Q { \c$ds_app_arg_21 [37:35] \c$ds_app_arg_21 [31:0] }
  end
  cell $not $auto$opt_dff.cc:248:make_patterns_logic$2411
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A $procmux$1580_CMP
    connect \Y $auto$opt_dff.cc:247:make_patterns_logic$2410
  end
  cell $reduce_and $auto$opt_dff.cc:273:make_patterns_logic$2413
    parameter \A_SIGNED 0
    parameter \A_WIDTH 2
    parameter \Y_WIDTH 1
    connect \A { $auto$opt_dff.cc:247:make_patterns_logic$2410 $auto$bugpoint.cc:309:simplify_something$1244 }
    connect \Y $auto$opt_dff.cc:272:make_patterns_logic$2412
  end
  attribute \full_case 1
  cell $pmux $procmux$1579
    parameter \S_WIDTH 2
    parameter \WIDTH 37
    connect \A { 5'xx100 $auto$bugpoint.cc:309:simplify_something$1313 [31:0] }
    connect \B 74'10110xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    connect \S { $procmux$1581_CMP $procmux$1580_CMP }
    connect \Y { \ds_3 [41:40] \ds_3 [37:35] \ds_3 [31:0] }
  end
  attribute \full_case 1
  cell $eq $procmux$1580_CMP0
    parameter \A_SIGNED 0
    parameter \A_WIDTH 2
    parameter \B_SIGNED 0
    parameter \B_WIDTH 2
    parameter \Y_WIDTH 1
    connect \A $auto$bugpoint.cc:309:simplify_something$1313 [33:32]
    connect \B 2'01
    connect \Y $procmux$1580_CMP
  end
  attribute \full_case 1
  cell $logic_not $procmux$1581_CMP0
    parameter \A_SIGNED 0
    parameter \A_WIDTH 2
    parameter \Y_WIDTH 1
    connect \A $auto$bugpoint.cc:309:simplify_something$1313 [33:32]
    connect \Y $procmux$1581_CMP
  end
  cell $mux $ternary$orange_crab.v:2664$382
    parameter \WIDTH 38
    connect \A 38'xx00xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    connect \B { \ds_3 [41:40] 1'1 \c$ds_case_alt_24 [37:35] \c$ds_case_alt_24 [31:0] }
    connect \S $auto$bugpoint.cc:309:simplify_something$1244
    connect \Y { \result_62 [42] \result_62 [39:35] \result_62 [31:0] }
  end
  cell $mux $ternary$orange_crab.v:2669$383
    parameter \WIDTH 35
    connect \A 35'0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    connect \B { 1'1 \ds_3 [36:35] \ds_3 [31:0] }
    connect \S \ds_3 [37]
    connect \Y { \c$ds_case_alt_24 [37:35] \c$ds_case_alt_24 [31:0] }
  end
  cell $mux $ternary$orange_crab.v:2801$408
    parameter \WIDTH 37
    connect \A $auto$bugpoint.cc:309:simplify_something$941
    connect \B { 4'1xxx \c$ds_app_arg_21 [31:0] 1'0 }
    connect \S \c$ds_app_arg_21 [37]
    connect \Y { $delete_wire$2834 $auto$bugpoint.cc:309:simplify_something$841 [26:0] $delete_wire$2833 }
  end
  connect $auto$bugpoint.cc:309:simplify_something$841 [27] 1'1
  connect $auto$bugpoint.cc:309:simplify_something$865 29'x
  connect $auto$bugpoint.cc:309:simplify_something$873 29'x
  connect $auto$bugpoint.cc:309:simplify_something$877 28'x
  connect $auto$bugpoint.cc:309:simplify_something$921 29'x
  connect \c$case_alt_50 { 1'x \result_62 [37:35] 3'x \result_62 [31:0] }
  connect \c$ds_app_arg_21 [34:32] 3'x
  connect \c$ds_case_alt_24 [34:32] 3'x
  connect { \ds_3 [39:38] \ds_3 [34:32] } { \ds_3 [40] \ds_3 [40] 3'x }
  connect { \result_62 [41:40] \result_62 [34:32] } { \result_62 [39] \result_62 [39] 3'x }
end
```
save as `bad.il` and call `yosys -p synth_ecp5 bad.il`.
Calling `rename -scramble-name`  before the synth makes the error disappear, which suggests there's a name collision (though I'm not sure why there are two separate instances of `$lut\c$case_alt_50[29]`)

KrystalDelusion, 2025-08-15:
After spending a few hours staring at this, I think using `splitnets` before `abc9` is far less of a "hack" than the current approach of recovering multi-bit signals in `read_aiger -wideports`.  #4349 shows that appending `[%d]` to the end of a signal name is not reliable as a means of indexing signals.  Here we get (as far as I can tell) `abc` nonsense with `case_alt_50[29]` being used to create a `$lut` cell before `case_alt_50` has been merged into a single port because of port ordering or something I guess (hence the transience under `rename -scramble-name`), which (as already mentioned) is resolved by using `splitnets`, and avoiding the ambiguity of wide ports.  I'm just not sure if it should be part of `abc9`, or if we add it to each `synth_*` flow, but probably the latter.

Ravenslofty, 2026-09-16:
I've tried all of these test cases again, and none of them crash with Yosys 0.68, so I think I'm going to close this for now.

Feel free to reopen if you are still having crashes.

closed 2026-09-16 by Ravenslofty