YosysHQ/yosys #4453: IDES4_MEM gowin module is missing

LaneaLucy, 2024-06-13:
### Version

Yosys 0.42+12 (git sha1 62bff3a20, g++ 11.4.0-1ubuntu1~22.04 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

try to synthesize the output of litex

to get the verilog file:
python3 -m litex_boards.targets.sipeed_tang_primer_20k --build

here a example yosys script
```
verilog_defaults -push
verilog_defaults -add -defer
read_verilog /home/lanealucy/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv.v
read_verilog /home/lanealucy/litex/build/sipeed_tang_primer_20k/gateware/sipeed_tang_primer_20k.v
verilog_defaults -pop
attrmap -tocase keep -imap keep="true" keep=1 -imap keep="false" keep=0 -remove keep=0

synth_gowin  -top sipeed_tang_primer_20k
write_json  sipeed_tang_primer_20k.json
``` 

### Expected Behavior

synthesize 

### Actual Behavior

```
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.42+12 (git sha1 62bff3a20, g++ 11.4.0-1ubuntu1~22.04 -fPIC -Os)

-- Executing script file `sipeed_tang_primer_20k.ys' --

1. Executing Verilog-2005 frontend: /home/lanealucy/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv.v
Parsing Verilog input from `/home/lanealucy/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv.v' to AST representation.
Storing AST representation for module `$abstract\VexRiscv'.
Storing AST representation for module `$abstract\DataCache'.
Storing AST representation for module `$abstract\InstructionCache'.
Successfully finished Verilog frontend.

2. Executing Verilog-2005 frontend: /home/lanealucy/litex/build/sipeed_tang_primer_20k/gateware/sipeed_tang_primer_20k.v
Parsing Verilog input from `/home/lanealucy/litex/build/sipeed_tang_primer_20k/gateware/sipeed_tang_primer_20k.v' to AST representation.
Storing AST representation for module `$abstract\sipeed_tang_primer_20k'.
Successfully finished Verilog frontend.

3. Executing ATTRMAP pass (move or copy attributes).

4. Executing SYNTH_GOWIN pass.

4.1. Executing Verilog-2005 frontend: /usr/local/bin/../share/yosys/gowin/cells_sim.v
Parsing Verilog input from `/usr/local/bin/../share/yosys/gowin/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\LUT1'.
Generating RTLIL representation for module `\LUT2'.
Generating RTLIL representation for module `\LUT3'.
Generating RTLIL representation for module `\LUT4'.
Generating RTLIL representation for module `\__APICULA_LUT5'.
Generating RTLIL representation for module `\__APICULA_LUT6'.
Generating RTLIL representation for module `\__APICULA_LUT7'.
Generating RTLIL representation for module `\__APICULA_LUT8'.
[... 155 lines cut ...]
Generating RTLIL representation for module `\FLASH64KZ'.
Generating RTLIL representation for module `\I3C'.
Generating RTLIL representation for module `\IODELAYA'.
Generating RTLIL representation for module `\IODELAYC'.
Generating RTLIL representation for module `\SPMI'.
Generating RTLIL representation for module `\IODELAYB'.
Generating RTLIL representation for module `\PLLO'.
Generating RTLIL representation for module `\DCCG'.
Generating RTLIL representation for module `\FLASH96KA'.
Generating RTLIL representation for module `\MIPI_DPHY_RX'.
Generating RTLIL representation for module `\CLKDIVG'.
Successfully finished Verilog frontend.

4.3. Executing HIERARCHY pass (managing design hierarchy).

4.4. Executing AST frontend in derive mode using pre-parsed AST for module `\sipeed_tang_primer_20k'.
Generating RTLIL representation for module `\sipeed_tang_primer_20k'.

4.4.1. Analyzing design hierarchy..
Top module:  \sipeed_tang_primer_20k

4.4.2. Executing AST frontend in derive mode using pre-parsed AST for module `\VexRiscv'.
Generating RTLIL representation for module `\VexRiscv'.
ERROR: Module `\IDES4_MEM' referenced in module `\sipeed_tang_primer_20k' in cell `\IDES4_MEM_15' is not part of the design.
```

whitequark, 2024-06-14:
> to get the verilog file (im running the command in "/home/lanealucy/litex/"):

Please attach the complete reproduction steps to this issue, including all files required to reproduce it without depending on third-party software. In this case this would be all of the Verilog files your script references.

LaneaLucy, 2024-06-14:
[sipeed_tang_primer_20k.zip](https://github.com/user-attachments/files/15840969/sipeed_tang_primer_20k.zip)

leviathanch, 2024-08-16:
Hi, I've got the exact same problem!

leviathanch, 2024-08-16:
There's a IDES4_MEM in /opt/gowin-1.9.8/IDE/bin/prim_syn.v but no IDES4_MEM_15
It might be an issue with Apypula respectively with Litex itself

leviathanch, 2024-08-16:
OK, it's IDES4_MEM which is missing, I've just checked in the Verilog code

//------------------------------------------------------------------------------
// Instance IDES4_MEM_15 of IDES4_MEM Module.
//------------------------------------------------------------------------------
IDES4_MEM IDES4_MEM_15(
    // Inputs.
    .CALIB (1'd0),
    .D     (gw2ddrphy_dq_i15),
    .FCLK  (sys2x_clk),
    .ICLK  (gw2ddrphy_dqsr901),
    .PCLK  (sys_clk),
    .RADDR (gw2ddrphy_rdpntr1),
    .RESET (sys_rst),
    .WADDR (gw2ddrphy_wrpntr1),
    
    // Outputs.
    .Q0    (gw2ddrphy_bitslip15_i[0]),
    .Q1    (gw2ddrphy_bitslip15_i[1]),
    .Q2    (gw2ddrphy_bitslip15_i[2]),
    .Q3    (gw2ddrphy_bitslip15_i[3])
);  

The issue is with Apycula not providing this cell as a dependency

leviathanch, 2024-08-16:
OK, no, seems to be a problem in Yosys after all

leviathanch, 2024-08-17:
I made a pull request solving this issue
https://github.com/YosysHQ/yosys/pull/4545