YosysHQ/yosys #4456: Feature Request: Add Support for Input Filelist -f Option

tarik-ibrahimovic, 2024-06-15:
### Feature Description

Currently, Yosys requires specifying each file individually in the command line or script for synthesis or formal analysis. This can be cumbersome for large projects with many files. I propose the addition of an input filelist option, where Yosys can read a file containing a list of Verilog source files to be included in the synthesis run. This feature would streamline the process, especially for larger projects or those with many dependencies.

chili-chips-ba, 2024-06-16:
... `-f <filelist>` option is typical for simulators. Verilator also has it.... which then naturally ties into Yosys Formal Verification side. Our [project](https://github.com/chili-chips-ba/openCologne) is currently using a simple script to bridge this gap from a straight filelist to a Makefile list.

Ravenslofty, 2024-06-30:
Well, the most literal interpretation of this isn't possible:

```
    -f frontend
        use the specified frontend for the input files on the command line
```

and I use `-f verilog` quite frequently, so I'm going to take the liberty of removing the specific flag suggestion.

whitequark, 2024-06-30:
It sounds to me that the functionality is already provided in `read -f` command, via Verific: https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/verific.html.

chili-chips-ba, 2024-06-30:
[Verific](https://www.verific.com) is indeed great, and also used within many proprietary tools, including Vivado. It is however not opensource. 

- Should then the reference to it be looked at as a suggestion that the `-f` support is made available only for the non-free Yosys tier?!

whitequark, 2024-06-30:
If someone implements the `read -f` functionality in the open source codebase (which any interested party can simply submit a PR for), they should take care to be compatible with the Verific syntax, since (as far as I can tell) that's where the syntax definition comes from. I.e. it is not merely a list of files but rather a specific language.

chili-chips-ba, 2024-06-30:
@Ravenslofty any suggestion on a better suggestion? (would `-F` make sense?)

Ravenslofty, 2024-06-30:
`-F` is available, yes.

I'd be happy to try to find a quick workaround. Is your intention to re-use this file for other commands, like to pass to a simulator?

chili-chips-ba, 2024-06-30:
Yes.

A typical part of the **Reuse Methodology** is to have one-and-only-one filelist that brings in all RTL files of the entire design, or an IP block. This filelist is then used for all downstream tools that process the design RTL, from Linter on. 

Note that `-f` is typically nested and hierarchical in the sense that the filelist that represents higher-level designs can `-f` into itself a list that represent an IP component, which may also `-f` a list of smaller building blocks...

JasonBrave, 2025-03-03:
Hi, I have created a PR to implement this feature https://github.com/YosysHQ/yosys/pull/4926