YosysHQ/yosys #4457: CologneChip synthesis fails silently (simulation discrepancies)

tarik-ibrahimovic, 2024-06-15:
### Version

Yosys 0.42+10 (git sha1 7f89a45ad, g++ 11.4.0-1ubuntu1~22.04 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Unzip this folder [uart_issue_yosys.zip](https://github.com/user-attachments/files/15855396/uart_issue_yosys.zip)
Prerequisites : `Verilator`, `iverilog`, `CologneChip tools`, `gtkwave`
Follow the CologneChip tool installation procedure (set CC_TOOL environment variable to cc-toolchain-linux/)
In the unzipped folder run:
```
cd 3.build
make hw_all
```
Then, to see RTL source simulation run:
```
cd 2.sim
make
```
To see synthesis netlist simulation, run:
```
cd 2.sim
make synth_sim
make wave_iverilog
```

### Expected Behavior

It is expected from synthesis simulation and RTL source simulation to send 2 bytes via `uart_tx`, every 1.6 ms. While RTL simulation does this, synthesis netlist does not. RTL simulation in a waveform viewer looks like shown in the picture, uart transmits then goes silent.
[image]

### Actual Behavior

After synthesis, by simulating the netlist, uart goes completely silent, with no warnings during synthesis. It has been tested that used sv2v works correctly as RTL simulation of generated Verilog files works as it should. 
[image]

This also results in wrong hardware being implemented and uart transmitting sometimes correctly, sometimes not. The behavior changes as hard-coded bytes do. For example 0xabcd is transmitted as 0x89ef, but 0xa0b1 transmits correctly, listed in the figures below.

All of this is happening while no synthesis errors or warnings are apparent. I have tried disabling the optimization of state machines on the suggestion of [this comment](https://github.com/chili-chips-ba/openCologne/issues/17#issuecomment-2170527516), but no big progress was made. 

This uart design has been proven good both on Xilinx and Gowin FPGAs using proprietary tools , noted in the discussion [here](https://github.com/chili-chips-ba/openCologne/issues/17#issuecomment-2170527516)

[image]
[image]

chili-chips-ba, 2024-06-17:
https://github.com/chili-chips-ba/openCologne/issues/17#issuecomment-2171829402

chili-chips-ba, 2024-07-06:
... from @pu-cc: https://github.com/pu-cc/yosys/tree/gatemate-regmap-fix

This bug is apparently also affecting `synth_xilinx`. On the positive side, `synth_ecp5` seems fine...