YosysHQ/yosys #4466: Unable to code FSM into binary format

titan73, 2024-06-20:
### Version

Yosys 0.42+10 (git sha1 ef9045882, clang++ 14.0.0-1ubuntu1.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

I"d like all FSM in my design coded  into binary format.
In my yosys script, I tried:
  1) synth -nofsm
  2) fsm -norecode (before en after synth command)
  3) fsm_recode -encoding binary (before en after synth command)
But no change. All FSM are coded into one-hot format

### Expected Behavior

All fsm 

### Actual Behavior

90.7.6. Executing FSM_RECODE pass (re-assigning FSM state encoding).
Recoding FSM $fsm$\current_state$32030' from module \stusb45_dpm_algo' using auto' encoding: mapping auto encoding to one-hot` for this FSM.
000 -> -------1
100 -> ------1-
010 -> -----1--
110 -> ----1---
001 -> ---1----
101 -> --1-----
011 -> -1------
111 -> 1-------

widlarizer, 2024-07-11:
Can you share the inputs so that we can reproduce this? Please check the offending FSM doesn't have the `fsm_encoding` attribute set or has it set to `auto` prior to the `fsm_recode` command being run

titan73, 2024-07-11:
I retested with synth -nofsm and I got it working. I don't know why it didn't work previously. I probably did something wrong prior to synth command. Let's close this issue then.

closed 2024-07-11 by titan73